opentx/opentx #4186: Horus OpenTX 2.2 N360 IT voice error

uragano47812, 2017-01-02:
Hello i see that on Horus with last N360 build and the correct SD card content there is a problem with IT language voice. For example in the timer if i have 5:51 the voice say 5 minutes and 5 second so it will read the decimal value as second and the not the real value in this case it would be fiftyone seconds. Also there are some error in the traduction like minutes instead of minuti and seconds instead of secondi.

I picked the csv file in the SD and i make my own traduction and created the language pack with TTSAutomate but the errors are still present even if now with the right spelling so seconds are now secondi.
Also on spalsh screen if italian is selected the horus hasn't sound instead in english it say welcome to opentx. 

If i use english voice there is no problem and the values are read in the right way.
I hope you understand the problem

3djc, 2017-01-02:
We do not have somebody that has volunteered to maintain Italian language, like we have for other languages.

The available sounds are here ; https://github.com/opentx/opentx/blob/next/radio/util/tts_it.py

And the language mechanics is here https://github.com/opentx/opentx/blob/next/radio/src/translations/tts_it.cpp

If a volunteers wants to pick it up, we will help !

uragano47812, 2017-01-02:
i modified tts_it.py but i can't modify tts_it.cpp as i don't understan all the logical.

Anyway the first problem to solve is to adjust timer reading as now it read only decimal seconds part of the timer.

es: 3:35 it will say three minutes and three seconds not thirty-five seconds (of course in italian)

uragano47812, 2017-04-28:
I uploaded the N370 with the right SDcard data but the problem on italian language is still there. As i said if timer is 4:58 for example the output voice say: 4 (quattro which is correct) minutes (instead of minuti) e (and) 4 (quattro instead of 58 fifty eight) seconds (instead of secondi).

If you play individual wav files of the numbers they are correct instead minute0.wav and minute1.wav as second0.wav and second1.wav are wrong and are like in english.

To me the wrong part is the logical to call the correct wav for numbers and some wav are still in english. I tried to use some custom wav done with TTSAutomate 2.8.2.6 but the numbers are not read anymore just mute only "minuti e secondi" in the output. I attach the generated files.

Let me know if i can help more
Thanks
[it.zip](https://github.com/opentx/opentx/files/964447/it.zip)

3djc, 2017-04-28:
We do not have any italian speaking developer, and nobody seems to have volunteered out of your post.

The first things that needs to happen is fix this file: https://github.com/opentx/opentx/blob/next/radio/util/tts_it.py

Indeed in there, minute0 (singular for minutes) is set too "minute", et plural to "minutes". If this is not right, we need somebody to review this file. That should take care of the english words.

If after that issue remains, then, something like the file bellow will need to be created some that we can understand the mechanics of the language (https://github.com/opentx/opentx/blob/next/radio/util/tts_doc_cz.md) . This is focused on unit and number because the issue for CZ was there. If for italian it is around time, then time need to be described in a similar way

uragano47812, 2017-05-02:
I checked again and fixed the "minute" and plural "minutes" voice generating new wav files. The problem that remain is that if i have time for example 6:10 it will always say "6 minuti" (right) and "6 secondi" wrong number for seconds it say always the minute value independent from which is the right seconds value if seconds are 10 or 55 it alway say "6" like minutes. the problem is probably on the call of the right wav file and it is in the cpp file but i'm not able to find the error because i don't understand the logic.
Let me know if i can help more.

3djc, 2017-05-02:
Then document exactly what the italian rules are for telling time in a form like the CZ document (doesn't require you to be a programmer)

PR #4889 (2017-05-06): Italian audio timers
- generate proper translation for hours minutes and seconds (this WILL require a sdcard version increment)

- Switch the time mechanic to english one, with italian words. It is likely not perfect, but probably usable, while current italian audio timers are not working at all

This fixes #4186 until further details are given
closed 2017-05-06 by bsongis