opentx/opentx #4190: Horus calling the texts  .txt for a model does not work

raso12, 2017-01-02:
Problem with OpenTx Horus

For each model, a note can be displayed at startup. This works only with the first model.
Next model no more

Raimund

3djc, 2017-01-02:
How did you call the file and where did you put it ?

raso12, 2017-01-02:
The file .txt is in the folder models.
Activated in menu model settings
Checklist. When selecting the model
The checklist should be shown

raso12, 2017-01-02:
For a model this works, not with all other models. Also not when the sender is restarted. The names are exactly the model names

3djc, 2017-01-03:
I does work, my guess is that you used model 'display' name and not model filename. You need to name it based on the name hightlighted in blue bellow:

[image]

raso12, 2017-01-03:
It only works if the name is for example
Model12. If the name is Thunderbird then for example
It does not work anymore. Also not if the file name
Model ... x.txt for which the model Thunderbird is.
Something's not right yet. What should I still try

For my X9d + works.

Raimund

3djc, 2017-01-03:
Please read again what I posted, especially WHERE to find the model name (and no, it cannot be Thunderbord, it WILL be something like modelXX.bin (replace .bin by .txt for your file)

raso12, 2017-01-03:
I found a solution.
I need to save the old models in for example model3.
Then call the .txt.
After that I can assign new names.
Then it works.

Example:
Model3.bin with the model name model3 calls model3.txt on then
Model3.bin with the model name Thunderbold calls Thunderbold.txt
is working

Since there is a conflict with the model names in the application and
The .bin and .txt file

I hope you can understand it synonymous

But is cumbersome. Its OK for me. Not for Opentx.

Many thanks for the best radioprogram ever....

Raimund (My English is bad...)

raso12, 2017-01-03:
Probably a memory location needs to be re-addressed

3djc, 2017-01-03:
We fixed it, but somehow broke it again by hte look of it, I will fix it again. In the next release, it will be the same as the .bin name (since you can give names to model you may not be able to give filemanes (special characters))
 Sorry for that

3djc, 2017-01-04:
We had a discussion about this yesterday, and here is our proposal:
- we will keep current way of doing ('similar to Taranis)
- in addition to that, you will also be able to use the modelxx.txt (even if your plane is called 'MySuperPlane')
- in case both are present (you should not be doing that, but just in case), the taranis style will have priority

projectkk2glider, 2017-01-05:
Just one comment. The usage of `.bin` file name will break when user will rearrange the model in Companion for example. The underlying `.bin` file might be renamed in that process. So using real model name, like on Taranis is better in this aspect also.

3djc, 2017-01-05:
Well, re-aranging should not imply .bin remane, since it can be dealt with within models.txt, I think we should avoid as much as we can renaming the bin file (but yes, there could be instances where this happens, so the PR gives you choice)

kilrah, 2017-01-05:
I also think companion should never touch the filenames and use the same methods as the radio to find a free one when needed.

3djc, 2017-01-12:
The mentioned change will be available in next nightly

PR #4195 (2017-01-08): 3djc/horus checklist improvement
- current way of working (like Taranis stays)
- in addition to that, you will also be able to use the modelxx.txt (even if your plane is called 'MySuperPlane')
- in case both are present (you should not be doing that, but just in case), the taranis style will have priority

This fixes #4190
closed 2017-01-12 by 3djc