opentx/opentx #4192: In 2.2RC8 Timer with Minute call enabled calls only number - no "minutes"

athertop, 2017-01-03:
In OTx 2.1.9 enabling the MInute call on a timer calls "n Minutes", but in 2.2 the "minutes" suffix is not called, so just a number is called 1,2,3 etc each minute. Can we have the 2.1 functionality restored please?

3djc, 2017-01-03:
Do you have the right sdcard content ?

athertop, 2017-01-03:
Actually I am running the 2.1 SD card content in the SIM - are you suggesting there is an additional wav in the 2.2 SD sound pack which is required for the "minutes" suffix?

kilrah, 2017-01-03:
The sound pack for 2.2 is different than the one for 2.1. All units have been renamed among others.

closed 2017-01-03 by kilrah

athertop, 2017-01-03:
Thanks kilrah - I will test my LUA with the new sound pack. Cheers for the info.

kilrah, 2017-01-03:
Lua scripts will also need adapting of course.

athertop, 2017-01-03:
any info provided as to the changes?

athertop, 2017-01-03:
tested with 2.2 sound pack and it does call the "...minutes" suffix correctly. Apologies and thanks for your time.