opentx/opentx #4198: Taranis X9D+ on OpenTX 2.2 N360: Sensor lost message (sensorko.wav) 10s after switch on

RipperDrone, 2017-01-04:
Taranis always sounds a "sensor lost" message right after switching it on - both in real use and in the Companion RC Simulator.

I have a telemetry screen configured, but tried to delete telemetry sensors line by line - didn't resolve the issue...

bsongis, 2017-01-04:
ID changed? a 3DR thing?

kilrah, 2017-01-04:
You shoudl be able to see which sensor it is on the telemetry page... would help to diagnose.

RipperDrone, 2017-01-04:
These are my Telem sensors: 
[image]

RipperDrone, 2017-01-04:
Strange - it seems I have to disable BOTH RSSI and RxBt sensors to get rid of the warning (I tried resetting their IDs to 0). Disabling just one of the two still results in sensor lost message. Any other sensors I can disable but still hearing sensor lost. Seems not only the sensor is lost here, I am too :-)

RipperDrone, 2017-01-04:
Double checked: Only F101 and F104 active - still "sensor lost message": 
[image]

kilrah, 2017-01-04:
It's "normal" in the simu, it sends one value when starting then stops, which causes the lost message to sound.
Should problably be removed one day.

RipperDrone, 2017-01-04:
Ok, then I have to do the physical Taranis test again over the next days. The message has occured just the same on the real Taranis shortly after switching it on with a quadcopter model being bound to it. Never has shown up on OTX2.1 so far, this is new with 2.2 N360. I will try to narrow it down further...

RipperDrone, 2017-01-04:
Anyway, I know the "telemetry lost" message for sure (when I switch the quadcopter off), but I have never heard the "sensor lost" message before...

schwabe, 2017-01-05:
sensor lost is new in 2.2. It happens if telemetry is still coming but one sensor stopped getting new values.

projectkk2glider, 2017-01-05:
I don't believe you actually disabled sensors by setting their ID to zero. The name of the sensor is used to detect if it is present-configured. 
```
bool TelemetrySensor::isAvailable() const
{
  return ZLEN(label) > 0;
}
```

RipperDrone, 2017-01-05:
Hmm, further tests done with 1 quad: So far the message did not come up anymore on physical Taranis. Have deleted A2, A4, SWR sensors completely, kept the rest. Will watch closely how to provoke the sensor lost message again...

RipperDrone, 2017-01-05:
@projectkk2glider: No calculated sensor in my setup that I'm aware of...

jwarburton, 2017-05-15:
just upgraded my X9D+ to OpenTX 2.2.0 RC18 (N371).  Seeing this issue also with repeat "Sensor Lost" notifications.  Not losing telemetry, just specific sensors like VFAS, RSSI, Acy coming from FrSky over smartport.  Was not seeing this issue before I upgraded the radio.  Error (sensor lost) messages do not seem to start until I throttle up and am in the air (dont get them on the ground when not under power).  Is this still a known issue?  Any fix?  thanks!

kilrah, 2017-05-15:
There must be a problem with your onboard setup, either a smart port cable that causes intermittent contact, an S.Port ID conflict or a receiver with faulty telemetry transmission.

Look at the telemetry setup page, you should see the *s blink at least about once per second for each sensor. If not something is wrong. When nothing is received for a sensor for more than 5 seconds you get the "sensor lost" warning.

3djc, 2017-05-15:
This is a new error message, but you should not get it if your sensors are sending data at their normal rate. You do not give a lot of details your setup, so hard to be more specific

jwarburton, 2017-05-15:
thanks for the feedback.  I see your points on faulty cable, hardware, etc - but the experience is so similar each time I power up the aircraft.  Telemetry comes right up, no issues when on the ground and not armed.  No issues when armed and not throttled up.  I've looked at the telemetry page and all the * are flashing (i'll check again their cadence).  Once I throttle up, then the 'sensor lost' warning comes and goes.  I've watched the Telemetry page and I see the metrics coming online and dropping off again.  No metric is lost and doesnt come back, they seem to 'bounce' in connectivity, and each disconnect gives me another 'sensor lost' warning.  Please let me know what specific details on setup you'd like.  Its an X9D+ with RC18 on it.  Connected to a FrSky X4R-SB.  Smart port connected to a Lumenier LuxV2 FC.  thanks again for the help!

jwarburton, 2017-05-16:
I have other quads, mostly running Lumenier Lux v1 or 2 flight controllers, connected to either XSRs or X4R-SBs.  I can try other setups to troubleshoot - let me know.  As an easy test tonight, I tried my X9D+ running RC18 that is connected to my Acrowhoop v2 flight controller (tiny whoop).  One of that FC's virtues for the tiny whoop is that it has integrated telemetry, stock, built into the board.  Its almost brand new too, only got it a few weeks ago...  And...  tested RC18 with it and got the same intermittent 'sensor lost' notifications on VFAS, RSSI, Acy, etc...  Never had any telemetry drops or anything the past 2 weeks running 2.1.6 (i think - whatever my last FW was).  I'd like to think that rules out the cabling etc as that is all factory and wasnt an issue prior to RC18.  thoughts?  Please let me know what troubleshooting steps I can take, but i think (and ive seen on facebook the same opinion) that this sensor lost is an active bug in the release candidate?   thanks in advance!

kilrah, 2017-05-16:
I's say it's an FC / FC cabling issue. 

> and wasnt an issue prior to RC18

As already said your prior version did not have this warning, so it was most likely already happening but you simply weren't made aware of it.

3djc, 2017-05-16:
As far as we know, there is no bug with this,. Please remember that what you are getting data not from an official sensor, but from software trying to emulate one, there could be issues in that code too, or some settings on your FC could create that too.

closed 2017-05-16 by kilrah

trintcole, 2017-05-24:
I'm also having this issue.  Got sensor lost just sitting on the bench.  Lux v2 with an xsr.  Switched to my spare X4r and no issues.  Updated firmware on xsr and seemed to fix the issue.  Using xsr now, took it on maiden flight.  Flew for about ten minutes then I started getting sensor lost messages again.   Sometimes I don't get them. 

Should I look at all the sensors on the telemetry page to make sure the asterisk is flashing when I hear the message? 

Any other ideas on a fix?

3djc, 2017-05-25:
As far as we know the issue is not on OpenTx side, so no idea on the fix. You can have a look at the telem page to try to detect what sensor is missing the blink, but if you hear the message, one did

hhoweson, 2017-06-07:
Me and 2 other I fly with (the only 2 others) are also getting this issue on opentx 2.2
Is there any way to disable it without deleting the sound file (disable it for specific models would be great)

trintcole, 2017-06-07:
I did a firmware udate on my recievers.  I have 3 xsr recievers.  One of them gives me the sensor lost alerts with the lux v2.   None of them give me the sensor lost with an xracer f303 v3.1.  Very strange results.  I put x4r firmare in them and the sensor lost is gone on all recievers, but the range is completely gone.

zukasmichael, 2017-06-25:
I bought 3 XSR and all of them have 'sensor lost'. had to update XSR to a new firmware and it's gone now. But I can't read PID's with a KISS LUT anymore. And I have occasional 'sensor lost' on X4R after 2.2 taranis x9d+ upgrade. not all sensors lost but like FUEL and so on... definitely Taranis firmware !!!

derpixeldan, 2017-07-17:
Switched from a X4R to XSR and now get the "sensor lost" message. Tried new wiring and second XSR. Same thing. It keeps bugging randomly. Looking at the sensors I see that after a while it won't receive `VFAS`, `Fuel` and `Curr` any longer. Though the sensor lost message seems to be related to `RSSI` and `RxBt` since it keeps bugging even without the `VFAS`, `Fuel` and `Curr` sensors.

schwabe, 2017-07-17:
@derpixeldan That sounds like a issue with the XSR instead of the X4R. For that unfortunately only FrSky can help you. 2.2 just points out the problem more prominently.

3djc, 2017-07-17:
You should be able to look in the logs what sensor(s) is causing the issue. Your case kinda shows the issue is not with OpenTx, since it works fine with X4R (and there is no receiver specific computing, actually, we cannot even know what receiver is being used). It could also be linked to actual setup issue with the XSR (receiver positioning vs interferences, antenna positioning,...)

hhoweson, 2017-07-17:
Out of interest, how many people here use rf1, apparently there is an issue with it sending packets to the xsr too quickly which can cause this

trintcole, 2017-07-17:
You need to update firmware on the xsr.

Trinity Colburn
Trinity@ciamarine.com
(850)291-2767

geofrancis, 2017-07-17:
I get this with both my arduino and teesy based Mavlink to smartport telemetry converters. I believe it's due to it not updating rssi often enough. An option to disable this sensor lost warning would be ideal.

schwabe, 2017-07-17:
@geofrancis the sensor lost alarm is 10s. So if you have no sensor reading for 10s you get the alarm. A frequenz of more than 1/10s should be achievable, even with slow telemetry.

geofrancis, 2017-07-18:
that time out might work with genuine frsky sensors, but some of my adapters have up to 28 emulated sensors and some dont get updated that often so i just get "sensor lost" warnings constantly. 

it needs to have either a configurable timout or an option to disable .

3djc, 2017-07-18:
10s is a lot of time, even for Arduino. I would rather check that something is not wrong with your code (my guess is there is, I have custom made sensors out of arduino, and I have the issue the way around, I tend to update too quickly ...)

geofrancis, 2017-07-18:
basically im using openlrsng-gitsly fork for my mavlink telemetry connection and its very slow, only 300 bytes/s its then fed into er9x mikes arduino mavlink to s.port adapter or athertops teensy based one. im pretty sure the slow updates are because of the slow mavlink connection so there is not much I can do about that.

http://openrcforums.com/forum/viewtopic.php?f=7&t=8396&hilit=sport&start=390#p117719
https://github.com/athertop/MavLink_FrSkySPort

3djc, 2017-07-18:
Use logs to identify :)

geofrancis, 2017-07-18:
thats not something you can do in the field without a laptop.

3djc, 2017-07-18:
Sure, but if you are in the field without laptop, you are unable to fix anything, so knowing which sensor is giving you trouble won't help much either, will it ?

geofrancis, 2017-07-18:
it would let me know if it was a critical sensor that has failed. your suggestion is if i get a sensor lost warning, pack up and go home?

kilrah, 2017-07-18:
Having sound files to say which is planned... one day
I don't see how telemetry at <10s refresh is useful. You'd probably want to change the number of sensors sent...

geofrancis, 2017-07-18:
the other issue is if your hear "sensor lost" repeatedly, you have no idea if its a single sensor glitching or multple sensors failing.

jwarburton, 2017-07-18:
I've upgraded to the Open TX 2.2.0 full release (5.30.17) and I'm still having issues.  From my May post, they started as soon as I upgraded to then RC18.  I see the "Sensor Lost" issue manifest across all my drones, connected to FrSky X4R-SBs & XSRs.  Even to FuriousFPV acrowhoop FCs that have their LR1000 unit in it.  Pretty much everything I fly.  I gave up - I went into the sound files and deleted sensor lost.  I still get the haptic warning but its better without hearing it all the time.

schwabe, 2017-07-18:
We could display an ! instead of a * for the sensors which ran into the 10s timeout. Would that help you?

I don't believe that disabling the sensor lost message is a good idea, you are only masking a real problem.

kilrah, 2017-07-18:
> I see the "Sensor Lost" issue manifest across all my drones, connected to FrSky X4R-SBs & XSRs.

You're aware that FrSky have posted receiver firmware updates to solve the issue with some FCs, right?

jwarburton, 2017-07-18:
@schwabe the problem is that I have multiple sensors 'failing' and going into 'sensor lost' alert.  Its not a question of if the alert in the UI is a * or !.  After discovering all sensors I tried to whittle it down to what I needed, which is really just VFAS and RSSI.  Still got the problem even when down to these 2 values.  I totally agree that killing the verbal alert is not the right way, it was just that my X9D was talking the entire flight and I couldn't hear my timer countdown, etc.  I still have the haptic response coming from the X9D.

jwarburton, 2017-07-18:
@kilrah good call.  I believe when I first commented on this post in May, those FWs were not out.  Looks like the new XSR firmware came out in June.  I guess that is my next step.  I just didnt want to start taking birds apart to get at my RX for update / was hoping an iteration of 2.2 would resolve the issue or allow a user to adjust the acceptable window of no data before going into sensor lost alarm.  Guess I'm taking some drones apart ;)  I found a good how to update video here FYI:  https://www.youtube.com/watch?v=VCpzlijHNVw

schwabe, 2017-07-18:
10s is already quite a large time. I don't think that making this adjustable to an even longer time would be benificial.

derpixeldan, 2017-07-18:
@schwabe @trintcole @3djc thanks for the rapid responses, it was indeed the firmware. I totally missed that there is a new one specifically for F3.

`Improve the compatibility between the S.Port and F3 FC.`
 http://www.frsky-rc.com/download/view.php?sort=Firmware&down=391&file=Firmware-XSR