opentx/opentx #4205: Horus 2.2 N360: SPORT packets sent by LUA are sent to random sensor IDs

raphaelcoeffic, 2017-01-06:
I traced the SPORT signal at the back of the Horus, and it seems that SPORT packets sent by LUA scripts (here S6R.lua) are sent to random sensor IDs. However, the S6R.lua sends its READ requests to sensor 0x1B:
```
local function telemetryRead(field)
  return sportTelemetryPush(0x1B, 0x30, 0x0C30, field)
end
```

First one is sent to 0x00:
[image]

and then 0xCB:
[image]

and on and on.... each time another ID (0x00 is there a couple of times).

raphaelcoeffic, 2017-01-06:
Same behaviour with "next" -> random IDs are used to send the values out.

[image]

RealTadango, 2017-01-07:
I confirm the issue and have the same issue on my Horus

raphaelcoeffic, 2017-01-07:
Was trying to make a list of the sequence used, and then I see this:
[image]

raphaelcoeffic, 2017-01-07:
And here another strange thing. Here is the global picture:
[image]

And now a little closer, the poll request:
[image]

and the reply from OTX almost 9ms later:
[image]

raphaelcoeffic, 2017-01-07:
If you ask me, there is something seriously broken here...

schwabe, 2017-01-07:
Where are these captured. On the sport pin of the TX or the sport PIN on a RX?

raphaelcoeffic, 2017-01-07:
@schwabe on the TX

RealTadango, 2017-01-08:
All S.Port push issues are solved if you use the Taranis telemetry driver.

RealTadango, 2017-01-08:
Confirmed to work. ID's are correct, speed is good. Works as expected :)

raphaelcoeffic, 2017-01-08:
Works on my side with Horus as well. I'll test with Taranis right away.

raphaelcoeffic, 2017-01-08:
Looks good on Taranis as well. Tested with S6R.lua and both scripts from Betaflight (SpMsp.lua & BFSetup.lua).

3djc, 2017-01-11:
Fixed in next nightly

commit 58382d8a (2017-01-09): [Horus] Fixes #4205 (#4210)

[Horus] Fixes #4205
closed 2017-01-11 by 3djc