opentx/opentx #4219: Multi Protocol Module - Flysky - cant change number of channels

robthomson, 2017-01-11:
When using the multi protocol module with the flysky protocol..

You cannot set the max channel numbers transmitted.  Its fixed at 12ch (if I remember from tests yesterday)

This can cause issues for some models / flight controllers that expect ppm to only have 8ch.

The furibee flysky module has this issue.  The  RX can bind.. but channel values override each other due to the 12ch being sent when 8 are expected.

kilrah, 2017-01-11:
That should be a multiprotocol module issue - it does not allow us to specify how many channels to transmit, the serial protocol to it is always 16CH and the module then decides what do do with them depending on selected protocol.

closed 2017-01-11 by kilrah

schwabe, 2017-01-11:
Flysky also only has 8ch, you probably talking about FS AFDS2A. My FS receiver also only outputs 8ch over the ppm port and not all 12, you need ibus or sbus for that.