opentx/opentx #4222: Custom 2 point curve overshoots values to select.

JapieG, 2017-01-11:
When i create a custom curve on Horus n360:

Type: Custom
Count: 2pts

When i want to select a value for the Y axis it overshoots the values when i try to select it with the rotary encoder.  I know it is unwanted to change x value on 2 point curve. But the way it works now is a bug i think :-)

projectkk2glider, 2017-01-12:
True, two point custom curve should not even exist. I wonder if the same bug is also on 2.1?

kellybyrd, 2017-01-27:
Is this something a new-to-the-project dev could pick up as his first bug? Or is there a bunch of history and context required here?

projectkk2glider, 2017-01-27:
Should be fairly easy. If you are luycky you just need to adjust the min value based on curve type here https://github.com/opentx/opentx/blob/next/radio/src/gui/480x272/model_curves.cpp#L135

kellybyrd, 2017-01-28:
Ok, I can repro this and I finally underatand @projectkk2glider's comment that "two point custom curve should not even exist." because you would solve that problem with a standard two point curve, like I do for a curve applied for volume control on a slider. Thanks for being patient, I'll dig in now.

projectkk2glider, 2017-01-28:
Go ahead with your idea.

kilrah, 2017-01-28:
Can be solved at editing time, when you change number of points to 2 type gets set to standard, when you change type to custom if number of points is set to 2 set it to 3. Ideally... do the same in companion :)

kellybyrd, 2017-01-28:
I've got a PR out for just skipping the X editing. This is what the current Taranis code does in the simulator, so I followed that example.

closed 2017-01-28 by bsongis