opentx/opentx #4223: Possible bug with affects servo output

JapieG, 2017-01-11:
I had my left flap moving a couple of times without touching the transmitter. 

Horus N360
rBus-10 with x8r and XM+ connected.
x8r on iXJT and XM+ on external XJT

Saw this happening a couple of times while working in the mixer screen. But then it also happened when i did nothing. So i suspected the rBus-10 to do something weird. Then i heard @realtadango had the same thing happening when using a powerbox redundant setup. So that rules out the rBus10.

In my case it affects only channel 6. I don't know if this is openTX related but wanted to post it here. By accident i captured the phenomenon on video. So here is a clip. Watch the flap. I am NOT touching the radio at the moment it happens!

https://www.youtube.com/watch?v=tjCOegiJYi4

3djc, 2017-01-11:
Likely not an opentx issue.  If it happens again and you can  see the variations in the radio channel monitor,  we will investigate then.

closed 2017-01-11 by 3djc

RealTadango, 2017-01-11:
I have seen similair servo twitches.......

kilrah, 2017-01-11:
You're both using signal switching devices - the problem has 99% chance of being there.

RealTadango, 2017-01-11:
Maybe in the s.bus stream but not the powerbox

RealTadango, 2017-01-11:
it occurs often when i change mixers ect

kilrah, 2017-01-11:
When editing things it's normal, you can very much create steps or transients when changing sources etc.

RealTadango, 2017-01-11:
True but i am not changing the channels that glitch?

kilrah, 2017-01-11:
Then who knows. If you can come with something reproductible with a single module-receiver setup we'll have a look, but until then since nobody has reported this on a "normal" setup we don't expect this to be an OpenTX problem.

Your redundant setups are just calling for this kind of issues. Remember that when using 2 modules the paths are NOT synced and it is totally possible for both receivers to put out different signals for the same channel at a given instant.

RealTadango, 2017-01-11:
That is not how a powerbox syncs :) That is very clean, but i agree, without a reproducable way this cannot be fixed.

projectkk2glider, 2017-01-12:
Source of this issue needs to be traced down by a simple method of elimination. When that proves that OpenTX can be the culprit we will investigate further.

My first ideas for you to try (one step at the time, notice when the issue dissappears):
* change servo
* remove power box
* use different power source
* change receiver to D series
* change radio