opentx/opentx #4224: Companion 2.2 problems

kilrah, 2017-01-12:
Priority issues:
- [ ]  Simulate radio from within companion with Horus selected still overwrites everything in the configured SD card directory instead of using a temp folder

- [ ] Horus: Switch warnings read wrong (Aup Boff Cmid Dup Eup Fdown Gup on radio gives all on, A mid, G mid, rest up in companion)

- [ ] Horus: External module/Multi not read

- [ ] Horus / Flight modes: Trim values not read

- [ ] Horus / Source list: Shift of 4 elements (SC read as SG, MAX read as CYC2, TR4 read as TR8

- [ ] Horus / Switch list: Shift of 22 elements somewhere (L2 read as L24, FM1 read as ---)

- [ ] All Radios / Flight modes: Trim "OFF" missing in dropdown

General issues:

- [ ] Selected model not showing / can't be changed

- [ ] Radio settings: Haptic settings should be enabled, haptic length missing

- [ ] Antenna selection for internal module missing

- [ ] Trainer settings missing

- [ ] Telemetry: Protocol appearing when it shouldn't

- [ ] RSSI alarms have the Orange/red dropdowns instead of low/critical

- [ ] Top bar settings should be hidden

- [ ] Telemetry screens should be hidden

Minor:

- [ ] Radio settings: Hardware->Serial port, ADC filter, BT missing

- [ ] Model setup: Image missing

- [ ] Drag and drop a .otx doesn't work to open it

kilrah, 2017-01-12:
No problem to report for Taranis except for the selected model thing

benoit35690, 2017-01-12:
I have added a separate issue 4225 dealing with the multiprotocol (2 issues actually)

projectkk2glider, 2017-01-13:
I will start on the first item *Simulate radio from within companion with Horus selected still overwrites everything in the configured SD card directory instead of using a temp folder*

mhotar, 2017-01-13:
There is nothing for me in your list?
[image]

kilrah, 2017-01-13:
Awesome!

bsongis, 2017-01-13:
wow, it's really nice!

3djc, 2017-01-13:
U Rule!

mpaperno, 2017-01-13:
Very nice!  :)    I'm going to work on the simulator UI so that we don't have do duplicate the whole form for the 5th time (would that be "quintuplicating?").

I'm having some trouble with the list identifying which part(s) is/are affected, or where to look... companion vs. simulator (and maybe simu or firmware also?).  I realize this was a quick list mostly for people who already know what it all means.  But any further clues may help :)

projectkk2glider, 2017-01-20:
I will take `Drag and drop a .otx doesn't work to open it`

3djc, 2017-01-26:
Added : Horus models cannot be moved from one category to another

projectkk2glider, 2017-01-27:
Added: `Show Splash Screen on Startup` option in general settings should be removed

projectkk2glider, 2017-01-28:
Added:
 -  Horus model list: cannot paste model into category (only on existing model).
 -  Horus model list: drag&drop of models between two model list does not work.

kilrah, 2017-01-29:
Some updates

projectkk2glider, 2017-01-30:
Added:  Horus model list: can't rename category (from #4339)

kposadowsky, 2017-01-30:
Tmp1 and Tmp2 are not simulated from csv.

kilrah, 2017-01-30:
Off topic. Only what's in the window is simulated.

Riffer, 2017-01-30:
Maybe I misunderstand this video where I see Tmp1 and Tmp2 values in the simulation are being played from the csv:

https://www.youtube.com/watch?v=mYgJ-Vn1h9M

kilrah, 2017-01-30:
Ah yes they are indeed. You must make sure the IDs are the same than what's configured in the model's telemetry. Again waaay off topic.

projectkk2glider, 2017-01-30:
Keeping so many issues in this one is getting a bit hard to track. For example it is difficult to easily see which sub issues were fixed in which release. I propose we don't add any more new issues here.

kellybyrd, 2017-02-01:
I was about to add something here,  the "duplicate" context menu option doesn't do anything. The function isn't defined in MdiChild. I'm working on it now let me know if it should be part of this or a new issue.

mpaperno, 2017-04-10:
Last item in list is done (drag/drop).  Woot!  Close?

bsongis, 2017-04-11:
Thanks for your help on the last open items!

closed 2017-04-11 by bsongis

kilrah, 2017-04-11:
👍