opentx/opentx #4225: companion 2.2 multiprotocol

benoit35690, 2017-01-12:
Companion version : 2.2.0 RC8N360
Configured for a Taranis X9D

-> bug 1 :  the Field "Channels is gray. Not able to select the correct number of channels. Or the number of channels is not correct depending on the protocol.
How to reproduce
In the "Editing model" page
In the "Setup" tab
In the "External Radio Module" section
a) select "DIY Multiprotocol Module" in the "Protocol" field
b) select "KN" in the "Multi Radio Protocol" field
c) select "WLToys" in the "SubType" field

-> bug 2 : multiprotocol selection not saved
How to reproduce
a) after having selected the multiprotocol options, save the model
b) close it
c) open the same model
d) look at the multiprotocol options : the fields are empty

kilrah, 2017-01-12:
1st bug isn't one, the protocol between radio and module is specified as always being 16 channels. 
The module then chooses to use them or not depending on protocol.

benoit35690, 2017-01-12:
OK but it is not always 16.
If you first select "DSM2" for "Protocol", then "DIY Multiprotocol Module", the value is not updated and stays at 6.

3djc, 2017-01-15:
I cannot reproduce that scenario on next

kilrah, 2017-01-15:
Yeah closing that - given the refactoring will change everything RC8 observations make no sense anymore, new testing is needed.

closed 2017-01-15 by kilrah