opentx/opentx #4227: Rename "Custom Scripts" to "Mixer Scripts"

RCdiy, 2017-01-12:
Wouldn't "Mixer Scripts" be beter than "Custom Scripts" as the screen title?

3djc, 2017-01-12:
No, because they can contain script that have nothing to do with mixers, ie  RB10 scripts

RCdiy, 2017-01-12:
What are RB10 scripts?
Anyway it would be beneficial to have the mixer scripts more intuitive. Telemetry scripts we go to the telemetry scripts tab. Mixer scripts we go to custom scripts? Not intuitive.

kilrah, 2017-01-12:
Then the proper way would be to rename "mixer scripts" to something else...

3djc, 2017-01-12:
This screen allow you run scripts that are custom to a model. They can fulfill a large variety of functions, but share in common to have no UI, unlike telem scripts, which are therefore separated

RCdiy, 2017-01-12:
Understood. Would be nice to have this all documented. I am willing to help. Documented for the "user" and documented for the "developer". I see a new Lua 2.2 document has started. Suggest adding a new Lua scripts to the new 2.2 Manual/Guide. Willing to help.

fred99, 2017-01-12:
A more intuitive wording would be "background scripts"

3djc, 2017-01-12:
Background running is also a feature of telemetry scripts

RCdiy, 2017-01-12:
If changed to "Custom Scripts" and "Background Scripts" then change folders from "/SCRIPTS/MIXES" to "/SCRIPTS/CUSTOM" and "/SCRIPTS/TELEMETRY" to "/SCRIPTS/BACKGROUND" (or /SCRIPTS/BKGRND is there is a character limit). 

background <--> LCD scripts???

kilrah, 2017-01-12:
No, the point is that telemtry scripts can also do stuff in the background, so calling mixer/custom scripts "background" would be ambiguous.

3djc, 2017-01-12:
Given the nightmare created by sd content changes,  that is probably to be avoided

kilrah, 2017-01-12:
could be "control scripts"...

projectkk2glider, 2017-01-12:
We have three options: rename scripts or rename screen or do nothing. Lets look at what each change entails:

## 1. Rename `mixer scripts` -> `custom scripts`
* update Lua docs
* change SD card path (I prefer not to do this)
* explain to confused users that the path and name changed :angry: 

## 2. Rename screen title
* change the screen title

## 3. Do nothing
* no work :smile: 

Lets also remember how the name `mixer scripts` came about. The original intent of these scripts was to have some script outputs which are calculated by the Lua script based on some inputs. This is exactly like the mixer lines are working - taking input and producing output. We now see that mixer scripts can also be used for different purposes, but their basic interface is still input - output. 

The name `mixer scripts` is now well established and known so my preference is either option 2 or 3.

RCdiy, 2017-01-13:
Regardless lets at least get the documentation done. I am offering to help. Documentation is easy to change. Lets add a Scripts section to the 2.2 manual/user guide now. I will help.

projectkk2glider, 2017-01-13:
@RCdiy go ahead with the documentation. You can create issues over at the lua doc github if you need to discuss what you are going to do or just prepare a pull request with your proposed changes and we will review it.

RCdiy, 2017-01-13:
Okay.
May be this needs a new thread but....
I've written a hold mixer script. Works great in the simulator.
In companion I can't create a mix with inputs from my hold script outputs.
If I go to the radio and create the mix what happens when I read the model into companion and then write is back. Will the mix get messed up?

projectkk2glider, 2017-01-13:
The `LUANm` mixer sources represent outputs from the mixer scripts. `N` goes from 1 to 7, 1 meaning first mixer script. `m` goes from a to f, a means first output. So using `LUA3b` means second output of third mixer script.

This kind of questions could be discussed over on https://opentx.rocket.chat/channel/OpenTX_General

closed 2020-04-06 by stale[bot]