opentx/opentx #4234: Taranis x9D Opentx "Logical" switch function move to Global ability.

smartbird34, 2017-01-14:
I want
LOGICAL SWITCHES  currently page 10/12 of MODEL SETUP.  Created in "Radio Setup" screens for Global use. 

Equipment: 
TARANIS X9D Plus

VERSION
 FW: opentx-taranis -plus
VERS: 2.1.8
Date 2014-04-14
EEPR:217

Currently on Page 3/9 of RADIO SETUP aka Global is limited to hard wired switches and mapped 6 position pot.  Giving current programming of: 

GF1 ON   Volume    S1
GF2 SF^  BgmusicII
GF3  SF1  Bgmusic  airwolf
GF4  SF2  Bgmusic  brlstlv  "Bread" Lost in Love.. 
GF5  SF3  Bgmusic  breadif
GF6>>GF9   Bgmusic 3 more sound files. 

I want
LOGICAL SWITCHES  currently page 10/12 of MODEL SETUP.  Created in "Radio Setup" screens for Global use.  

So I can get on Page 3/9 of RADIO Setup = Global
GF1 ON   Volume    S1
GF2 SF^  BgmusicII
GF3  GL1  Bgmusic  airwolf
GF4  GL2  Bgmusic  brlstlv  "Bread" Lost in Love.. 
GF5  GL3  Bgmusic  breadif
GF6>>GF30 = GL4 >> GL26  Bgmusic 21 more sound files.

Reason to have this option is using a 6 position pot switch combined with a 3 position switch in a Logic table gives = 24 logic switch positions.  

 I want to  24 songs background in all aircraft.  Without using 24 L1 >> L24 and 24 SF1 thru SF24 per Model setup.    This will aid keeping them organized exchange of music files of slightly descriptive titles.

kilrah, 2017-01-15:
People use Lua scripts for that, there's a "Jukebox" one around.

kilrah, 2017-11-01:
#2409

closed 2017-11-01 by kilrah