opentx/opentx #4264: Improve cross-radio conversions

kilrah, 2017-01-21:
Some examples:

X9D->Horus: 
- Pot remapping can be done with some sense

Horus->Taranis:
- Throttle source shift
- Trim selections become mangled due to the 2 extra trims

In general when conversions are not straightforward or result in an invalid selection we could choose a default value and notify the user of what was changed in a messagebox or text file "conversion report"

kilrah, 2017-03-24:
Examples in #4570

raphaelcoeffic, 2019-08-14:
@kilrah #5534 has already provided some improvements (at least it is linked to this issue). Should we still leave this here open?

kilrah, 2019-08-14:
#5534 added a mechanism to do conversions from radio A to radio B, this issue was about doing something useful with it

raphaelcoeffic, 2019-08-14:
> this issue was about doing something useful with it

Thx. Do you think this is something that could be done for 2.3? (I mean, someone you know will have time do it in the very next future?)

kilrah, 2019-08-14:
It is something we should care about.

raphaelcoeffic, 2019-08-14:
> It is something we should care about.

Does that mean you think 2.3 should be rather delayed than being released without this feature? Just trying to figure out if that can be moved to another release.

closed 2020-06-06 by stale[bot]