opentx/opentx #4266: TX-VOLTAGE

erwabo, 2017-01-22:
I last saw this was missing in 2.1.7 and was supposed to be back in 2.1.8. I just recently got 2.1.9 but still did not see this function. Latest updates I saw was from January of 2016 so I was just wondering if this was something planned for the next build.

Thanks

kilrah, 2017-01-22:
Could you maybe detail what you're actually talking of?

erwabo, 2017-01-22:
When I use the "tx-voltage" channel in my lua script,  I'm not getting reading. There was a previous issue opened for this and it was stated in 2.1.7 it was missing and would be back in 2.1.8 but I have 2.1.9 and I still don't see it available.

projectkk2glider, 2017-01-22:
I will verify this, but nothing changed and it should work.

projectkk2glider, 2017-01-22:
Works here

```lua
local function run()
  local v = getValue("tx-voltage")
  print ("tx-voltage: "..v)
  local r = getValue("rssi")
  print ("rssi: "..r)
  lcd.clear()
  lcd.drawText(10,30, "tx-voltage: "..v.." rssi: "..r)
  return  0
end

return { run=run }
```
[image]

erwabo, 2017-01-22:
I must have something wrong in my script...I tried  a blank lua script and it worked. Here is the one I cant get it working with. I tried putting it in different places as well.

--[[ GLow Fuel - Fuel Guage
 by Eric Bojonell
 Version 1.0 - 20 January 2017

]]

local function getTelemetryId(name)
 field = getFieldInfo(name)
 if field then
   return field.id
 else
   return -1
 end
end

local function getValueOrDefault(value)
 local tmp = getValue(value)

 if tmp == nil then
  return 0
 end

 return tmp
end

local function init()
 ver = getVersion()
 if ver>="2.1.0" then
  A2 = getTelemetryId("A2")
  A1 = getTelemetryId("A1")

  else
  A2 = getFieldInfo(A2).id
  A1 = getFieldInfo(A1).id

 end

end

local function run()
  local v = getValue("tx-voltage")
  print ("tx-voltage: "..v)
    lcd.drawText(165,56, "TX: "..v)

end

local function run(event)

----------------------------------------
-- GAUGE VARIBLES;
----------------------------------------

 local x1 = 37
 local y1 = 40
 local angle
 local x2
 local y2
 lcd.clear()

----------------------------------------
-- LINES;
----------------------------------------

 lcd.drawLine(67,8,67,58, SOLID, 0)
 lcd.drawLine(130,8,130,58, SOLID, 0)

----------------------------------------
-- TIMER;
----------------------------------------

 local timer = model.getTimer(0)
  lcd.drawTimer(95, 5, timer.value, MIDSIZE)
  lcd.drawPixmap(75, 6, "/SCRIPTS/TELEMETRY/OLIMETRY/timer.bmp")

  ----------------------------------------
-- CLOCK;
----------------------------------------

  lcd.drawTimer(95, 23, getValueOrDefault(190), LEFT+MIDSIZE)
  lcd.drawPixmap(75, 23, "/SCRIPTS/TELEMETRY/OLIMETRY/clock.bmp")

  ----------------------------------------
-- RATE MODE;
----------------------------------------

  lcd.drawPixmap(75, 44, "/SCRIPTS/TELEMETRY/OLIMETRY/fm.bmp")
  if getValueOrDefault(94) > 0 and getValueOrDefault(97) < 0 then
   lcd.drawText(97, 48, "LOW", SMLSIZE)
  else
   if getValueOrDefault(97) == 0 and getValueOrDefault(97) < 0 then
    lcd.drawText(97, 48, "MED", SMLSIZE)
   else
    if getValueOrDefault(94) < 0 and getValueOrDefault(97) < 0 then
     lcd.drawText(97, 48, "HIGH", SMLSIZE)
    else
     lcd.drawText(97, 48, "MAN", SMLSIZE)
    end
 end
 end
----------------------------------------
-- FUEL;
----------------------------------------
--lcd.drawText(165, 55, "FUEL", SMLSIZE)
lcd.drawPixmap(135,2, "/SCRIPTS/TELEMETRY/OLIMETRY/fuel-gauge.bmp")

 x1=172
 for i=0, 10  do

  angle = math.rad(((17*i)-180))
  x2 = 25 * math.cos(angle) + x1
  y2 = 25 * math.sin(angle) + y1

 end

 local FuelVoltage = getValueOrDefault("A2")--get vspeed in m/s

-- Draw indicator

angle = math.rad((FuelVoltage*17)-180)
 x2 = 27 * math.cos(angle) + x1
 y2 = 27 * math.sin(angle) + y1
 lcd.drawLine(x1, y1, x2, y2, SOLID, 0)--GREY_DEFAULT)

----------------------------------------
-- RSSI;
----------------------------------------
local percent = 0

local CurrentRSSI = getValueOrDefault("RSSI")

  if CurrentRSSI > 90 then
    lcd.drawPixmap(35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh10.bmp")
  elseif CurrentRSSI > 80 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh09.bmp")
  elseif CurrentRSSI > 70 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh08.bmp")
  elseif CurrentRSSI > 60 then
  lcd.drawPixmap( 35, 1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh07.bmp")
  elseif CurrentRSSI > 50 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh06.bmp")
  elseif CurrentRSSI > 40 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh05.bmp")
  elseif CurrentRSSI > 30 then
  lcd.drawPixmap( 35, 1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh04.bmp")
  elseif CurrentRSSI > 20 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh03.bmp")
  elseif CurrentRSSI > 10 then
  lcd.drawPixmap( 35,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh02.bmp")
  elseif CurrentRSSI > 0 then
  lcd.drawPixmap(35, 1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh01.bmp")
  else
    lcd.drawPixmap(  35, 1, "/SCRIPTS/TELEMETRY/OLIMETRY/RSSIh07.bmp")

  end

  lcd.drawChannel( 46,  55, "RSSI",  SMLSIZE)

----------------------------------------
-- Battery;
----------------------------------------

   local myCurrent = getValueOrDefault("A1")
   local myMaxV = 5.2
   local myMinV = 4.0
   local myRangeV = myMaxV - myMinV
   local myAvailV = myCurrent - myMinV
   local myPercent = math.floor(myAvailV / myRangeV * 100)

   if myAvailV < 0 then myAvailV = 0 end
   if myPercent < 0 then myPercent = 0 end
   if myPercent > 100 then myPercent = 100 end

   local myPxHeight = math.floor(myPercent * 0.37)
   local myPxY = 11 + 37 - myPxHeight

   lcd.drawPixmap(10,  1, "/SCRIPTS/TELEMETRY/OLIMETRY/batteryh.bmp")

  if myPercent > 0 then
     lcd.drawFilledRectangle(13, myPxY, 09, myPxHeight, FILL_WHITE )
   end

   local i = 36
   while (i > 0) do
     lcd.drawLine( 14, 10 + i,  20,  10 + i, SOLID, GREY_DEFAULT)
     i= i-2
   end

   if (myCurrent > myMaxV) or (myCurrent < myMinV) then
  lcd.drawChannel( 10, 55, "A1" ,PREC1 + LEFT + SMLSIZE)
   else
  lcd.drawChannel( 10,55, "A1" ,PREC1 + LEFT + SMLSIZE)
   end

 end
 return { run=run, init=init}

Works here

[snapshot_01]<https://cloud.githubusercontent.com/assets/5950438/22182690/f50d54c0-e0ab-11e6-9814-dc5cdf7f8221.png>

[https://cloud.githubusercontent.com/assets/5950438/22182690/f50d54c0-e0ab-11e6-9814-dc5cdf7f8221.png]

-

projectkk2glider, 2017-01-22:
Please try to get help for this on forums. Looks like a bug in your script and not OpenTX. If you however determine that it is out bug, then reopen this issue.

closed 2017-01-22 by projectkk2glider