opentx/opentx #4277: Q7 Battery Graph not correct

robthomson, 2017-01-24:
Small display issue on the Q7 batter graph.

Essentially despite the  voltage range being correct - and the battery fully charged (checked with multimeter) the graph never shows the graph as 'full'.

I suspect the issue is that the voltage is not reading the calibrated voltage value - but rather the raw and incorrect voltage.

3djc, 2017-01-24:
Please give data : what voltage you measured, what was displayed, what is you range

robthomson, 2017-01-24:
Lipo: 2 cell
Graph: 3 lines (about 25%)
Displayed voltage:  7.7v
Voltage Range: 6.6-8.5

3djc, 2017-01-24:
Not sure I understand, you said full charged bat, but surely this is not the case at 7.7 ?

robthomson, 2017-01-24:
No.. its not currently charged.  But 7.7 is not 25%.

Closer to 57% if my maths is right?

kilrah, 2017-01-24:
The charge of a lipo is not linear - the meter however is because it's not tuned to any particular type. 
It's only intended as a very rough estimation.
Also - Your lipo won't ever reach 8.5V, but the battery meter will only show full there as that's what you've configured. Typically if you want the graph to be useful you set the end points to a bit LESS than your battery's actual range, e.g. 6.8-8.2

robthomson, 2017-01-24:
Ok.. let me clarify further.

The pack is on 7.7 now.

But the bar graph still only showed 3 bars even when it was a fresh pack at 8.2!

projectkk2glider, 2017-01-24:
Your math is right `100×(7,7−6,6)÷(8,5−6,6) = 57%`

I also tested on simulator, looks ok:
[image]
[image]

What is displayed voltage in your case?

projectkk2glider, 2017-01-24:
Perhaps it was fixed already, which version of OpenTX are you using?

robthomson, 2017-01-24:
Again...  let me show you more values.

 changed mine to a LiFe

Pack is on 6.6v
Range: 6.2v->7v

Graph still shows only 3 bars.

Running: 2.2.0N360 on the x7d

Rob

robthomson, 2017-01-24:
If I change the range to 6.2 -> 6.6v

I get 5 bars.  50%

projectkk2glider, 2017-01-24:
>Pack is on 6.6v

Is this measured or displayed voltage?

robthomson, 2017-01-24:
Measured and Displayed

I think the issue is that the voltage for the graph is maybe coming from the non calibrated value?

My radio read this value as higher when I plugged a meter on the battery with the radio running.

So I reduced it to match the meter in the calibration/setup screens.

projectkk2glider, 2017-01-24:
Ok, fund fbe11119b8e7257c49b61ac1f72acecf32a150cd which fixed this, which is probbably after the N360

3djc, 2017-01-24:
Yes N360 was Dec 4th, this is 11th

projectkk2glider, 2017-01-24:
Jep, N360 was 3. December. Case closed

closed 2017-01-24 by projectkk2glider

robthomson, 2017-01-24:
Hmm..    I downloaded and installed this firmware from companion a matter of a week ago! 

Must work out why it is an old version!

3djc, 2017-01-24:
I believe x7 are factory delivered with a firmware dated Dec 22nd,  wich is newer than what you can currently get from companion updates. That will change when we release n361

robthomson, 2017-01-24:
OK.    Makes sense.   My re-flash to get multi module support downgraded me :)