opentx/opentx #4278: Multi Protocol Module - power cycle bay when swapping module

robthomson, 2017-01-24:
Occationally when using a multi protocol module I notice and issue in which the protocol does not change when swapping between two multi protocol module aircraft.

eg.  Flysky & DSMX

The solution is always to powercycle the radio.

I suspect the issue is a simple fix of powercycling the module bay when the models get swapped?

projectkk2glider, 2017-01-24:
Which radio and OpenTX version?

robthomson, 2017-01-24:
Has happened to me on horus and q7.  Latest firmware downloaded from companion :-)

schwabe, 2017-01-24:
Occationally when using a multi protocol module I notice and issue in which the protocol does not change when swapping between two multi protocol module aircraft.

How do you notice that? Does your model not work or does the new model not react?

robthomson, 2017-01-24:
New model does not bind.

Its as if the module is still stuck on the old protocol.

3djc, 2017-01-24:
Is the red led turned on when that happens ?

bsongis, 2017-02-14:
Duplicate of #4404 which has been fixed yesterday
Well in fact #4404 is duplicate of this one, but it doesn't matter

closed 2017-02-14 by bsongis