opentx/opentx #4279: Suggestion - Change app name on Mac from "companion" to "OpenTX Companion"

DIYglenn, 2017-01-24:
It's confusing when it's been a while, and I want to open OpenTX companion, so I search for "open" on my Mac, but nothing. The app is just called "companion".

Also, it doesn't have the version attribute set, so you cannot see which version you have installed.

schwabe, 2017-01-24:
[image]

In OpenTX 2.2 this already done. I am not touching the cmake 2.1 build script

schwabe, 2017-01-24:
Renaming of 2.1 binary will probably not happen (I suggest to rename it locally) and for 2.2 it is already fixed, so I will close this as fixed in 2.2 and wontfix in 2.1

closed 2017-01-24 by schwabe

DIYglenn, 2017-01-24:
Hey, that looks great! 
Thanks, I did not know.