opentx/opentx #4285: HORUS - allow easy switching of antenna port

dk7xe, 2017-01-25:
The used antenna on the HORUS (internal / external) is part of the model settings.
But i think it would be handy to have an more easy way to switch between internal and external.
Because there is a high risc that users are configuring to use as default the external antenna but they forget to attach one (and forgot that they have configured to use external).

I suggest to have as default, when powering on, always the internal antenna and have a simple menu to switch between internal and external antenna. e.g. include it in the pop up menu for Model Select
[image] 

An advanced feature besides this would be that the VSWR is checked and if external antenna is configured to automatically switch to internal antenna on bad VSWR.

An icon should show the actual configured antenna.

schwabe, 2017-01-25:
VSWR is only an indication. It is not reliable. You should not depend on it to detect no external antenna attached.

3djc, 2017-01-25:
You easily set a model notes reminding you to connect the antenna

projectkk2glider, 2017-01-25:
Last time I loaded a model that had external antenna configured and radio was without it, I got a bad antenna warning immediately.

dk7xe, 2017-01-25:
I agree there could be notifications and warning will notify. But it would be more user friendly to allow antenna swithing via a simple menu rather than going into model and all the way down to the tx module section.

schwabe, 2017-01-25:
I am trying to understand the need for that. When you would regulary attach/detach an external antenna?

3djc, 2017-01-25:
Most models do not require external antenna.  If you have a model that requires it,  then you set it and you must have it.

dk7xe, 2017-01-25:
Personally i would mount the antenna based on the area i'm flying in and not the model. Therefore for me it makes no sense to have that based on the model. If i'm flying in the wood, open field (longer range) - external. If i'm flying on a small race track, or indoor (or just forgot to pack the external ant into the backpack) - use internal.

RealTadango, 2017-01-26:
I think it should be a general option with the following options:
Force internal
Force internal and external
Per Model

Or just simply remove the model antenna choice and make it a general only (my choice :))

Adding and removing the antenna is bad for the connector and is really the worst option and should not be encouraged.

dk7xe, 2017-01-26:
Most datasheets i have seen on SMA gurantee a durability of 500 mating cycles, btw. Should be enough for a TX lifetime if i mount once in a while an external antenne. ;)

dk7xe, 2019-08-21:
No update?

bsongis, 2019-08-21:
This feature makes sense to me. Inline with what is done in ACCESS

3djc, 2019-08-21:
No quite, this requires to activate the antenna at each model load or change

kilrah, 2019-08-21:
Would have to have a new setting "Ask" in the antenna choices. If that bit is set then on model load you get a choice populated with the current selection.

bsongis, 2019-08-21:
@3djc I don't get your comment. Today this feature would be only applicable to ACCST (because ACCESS = global for the whole radio)

bsongis, 2019-08-21:
I do agree with the additional "Ask" setting.

3djc, 2019-08-21:
What I mean is that on each model load/change, you would have to select the antenna, that's not access behavior

bsongis, 2019-08-21:
Ok. Then yes, if the user chooses "Ask" in general settings, yes at each model load there would be a question (internal / external antenna)

kilrah, 2019-08-21:
I was thinking of "Ask" as a model setting, in the same list as the current choice. So displayed as internal, external, ask

bsongis, 2019-08-21:
Understood, it means:

**RADIO**
1) Internal
2) External
3) Ask
4) Per model

**MODEL** (if per model)
1) Internal
2) External
3) Ask

kilrah, 2019-08-21:
OK both works too :)

bsongis, 2019-08-22:
Random comments:
- the confirmation dialog has been removed (still some strings remaining if we really decide to remove it)

bsongis, 2019-08-22:
@mhotar what about adding an icon on top of the RSSI bars on Horus (when the external antenna is used) ?
[image]

mhotar, 2019-08-24:
What would you like to display in case both antennas are used?
[image]
I would say, the first colums is better because of the icon size (16x16), but i do not know how good the X10 LCD is. Let me know which set would you like in the png format.

dk7xe, 2019-09-03:
@mhotar i agree the left column is my preference as well.

dk7xe, 2019-09-03:
current funtionality is that Antenna setting is on model level.
To me that's sufficiant because i think there will be always an exception to the default that would require "Per model" on RADIO level.

Please do not forget the both option at least on MODEL and as selection when ASKed

MODEL (if per model)

1. 
2.     Internal
3.     External
4.     Both
5.     Ask

3djc, 2019-09-03:
Both doesn't mean anything on XJT, it is either int or ext. How it deals with that depends on exact xjt module, but there is no both option

bsongis, 2019-09-03:
Thanks @mhotar for the proposal, I just see it now! I do like the left bottom icon, I will take it (I need only one icon, there is no BOTH possibility)

bsongis, 2019-09-03:
@mhotar I would need the png transparent icon, thanks again!

kilrah, 2019-09-03:
Well the point is that on the X12 it's both, and on the x10 it's ext only, so yes it kinda makes sense to use a different icon for each

3djc, 2019-09-03:
Then we also need to change the string to match int/ext/ask or int/both/ask depending on radio

bsongis, 2019-09-04:
Agreed. I didn't have in mind that it was different on X12 / X10

bsongis, 2019-09-05:
@mhotar I will commit a small icon which is far from perfect!

closed 2019-09-05 by bsongis