opentx/opentx #4292: OPEN TX 2.2N361 with Taranis X9D Plus Problem

spadeadam, 2017-01-26:
Hi, I have just downloaded and installed the latest TX Companion for Windows (2.2N361) and SD Card (x9-2.2V0005) and updated my Taranis X9D+ using Companion. When I switch the Taranis on I get a blank screen and the Taranis vibrates. When I revert to 2.2N360 all is well again.

projectkk2glider, 2017-01-26:
Are **you sure** you have `Taranis Plus` selected in the radio profile and not `Taranis`?

projectkk2glider, 2017-01-26:
Also what is the name of the downloaded firmware file?

spadeadam, 2017-01-26:
Are you sure you have  Taranis Plus  selected in the radio profile and not  Taranis ?
Yes, I have double checked that Taranis Plus is selected.

Also what is the name of the downloaded firmware file?
opentx-x9d+-noheli-lua-en-2.2.0N361.bin

Dutchy-79, 2017-01-26:
Same problem with my eu lbt Taranis X9d plus 

Just updated from RC8 to RC9, when turning on I get blank screen and vibration goes crazy.
Taranis X9D Plus using this file opentx-x9d+-noheli-lua-massstorage-eu-multimodule-en-2.2.0N361.bin

Edit, back to RC8 was no problem, works fine only thing I can't is make Amber say welcome to Taranis instead off open tx

Copied from my rcgroups.com message to plaisthos. He sent me to this thread.

projectkk2glider, 2017-01-26:
@Dutchy-79 and @spadeadam can you try re-downloading firmware and flashing again?

schwabe, 2017-01-26:
opentx-x9d+-noheli-lua-massstorage-eu-multimodule-en-2.2.0N361.bin works fine on my remote

Dutchy-79, 2017-01-26:
I will, but I'm back home in 14 hrs

spadeadam, 2017-01-26:
I will retry the upload again tomorrow morning and report back my findings

powercycle, 2017-01-27:
This works as expected for me as well.

projectkk2glider, 2017-01-27:
We also tested and it worked. I must have been a glitch in early compilations. Reopen if needed.

closed 2017-01-27 by projectkk2glider

Dutchy-79, 2017-01-27:
Confirmed working from my side also. Thanks for the quick fix guys.

spadeadam, 2017-01-27:
Also working for me now....I see that the file size has changed from the original one I flashed my Taranis with yesterday. Many thanks for the prompt response.