opentx/opentx #4294: HORUS - add MP3 Player

dk7xe, 2017-01-26:
FrSkyOS on Horus is having a MP3 Player.
Please add this functionality to OpenTX as well.
This is useful if you fly a coreography and like to keep tricks in sync with a certain background music.
Or just for having some fun at the field ;)

kilrah, 2017-01-26:
We use the sound chip differently from FrSky OS, so can not implement MP3 decoding. 

Just convert your files to WAV.

closed 2017-01-26 by kilrah

dk7xe, 2017-01-26:
Ok, if it is not possible at all to use MP3 i'd like to reduce the scope to:

Audio files to be stored as WAV on separate folder independent from language e.g. SD card /MUSIC folder.

Functions should be made available to

    start certain audio file.
    stop audio file.
    pause audio file / and restart from same point again.
    jump to next file.
    jump to previous file.

Thank you!