opentx/opentx #4306: OPEN TX 2.2N361 with Taranis X9D Plus Problems

spadeadam, 2017-01-27:
I have a couple of problems with the latest release (these were also present in the previous release 2.2N360 but I was not aware of this forum to report the problem) which may be due to me doing something wrong :(

a. I use a 6 way multifunction switch S3 (which works perfectly in version 2.1.9) and it is present when I add it to the hardware and calibrate it, but when I try to select S3 in the INPUTS screen it is not present....maybe I have missed something.

b. When I set up a logical switch to report on battery voltage I select Cels in V1 but I get Seconds as the unit in V2 (should be Volts)........this works correctly in ver 2.1.9

projectkk2glider, 2017-01-28:
Your problem *b* is the wrong sound pack. Use sound pack compatible with OpenTx 2.2 http://downloads.open-tx.org/2.2/nightly/sdcard/

projectkk2glider, 2017-01-28:
Using simulator and current `next` I can select S3 as a source for an Input, I can also select S31- S36 in Logical Switches. Not sure what is wrong here.

@spadeadam are  you sure you have configured S3 properly in hardware setup page - set it to `multipos` and the re-calibrate and when calibrating leave S3 in each position for at least 0.5 second.

spadeadam, 2017-01-28:
@spadeadam are you sure you have configured S3 properly in hardware setup page - set it to  multipos  and the re-calibrate and when calibrating leave S3 in each position for at least 0.5 second.

I re-calibrated the 6 way multipos switch leaving 0.5 seconds between each position and I can confirm that the switch (s3) is working as it should.....thank you
*************************************************************************************************
Your problem b is the wrong sound pack. Use sound pack compatible with OpenTx 2.2 http://downloads.open-tx.org/2.2/nightly/sdcard/

I am using the latest download which was copied to a new SD-Card and then inserted in to my Taranis X9D Plus (http://downloads.open-tx.org/2.2/nightly/sdcard/sdcard-taranis-x9-2.2V0005.zip) the readme on the SD-Card states 2.2V0005.
When I set up a logical switch using V1 set to Cels to announce my battery voltage I get minutes and/or seconds.
When I set up a logical switch using V1 set to RxBat to announce my receiver battery voltage I get Volts which is correct.

Many thank for the help with the S3 multipos switch

projectkk2glider, 2017-01-28:
Then check the settings of the `Cels` sensor (unit). Is the displayed unit correct (Volts)? Fi you still have problems, then upload your settings and post link here.

spadeadam, 2017-01-28:
The displayed unit is seconds (should be Volts)

[image]

spadeadam, 2017-01-28:
Problem now sorted......I did not rediscover sensors after copying my model from a different Taranis.  Once discovered Cels the units were displayed as Volts.

Many thanks for your help

kilrah, 2017-01-28:
Then there is a conversion error somewhere that could be fixed, would be good to say which versions your source and destination taranis are running.

kroral, 2018-03-18:
If you have the onboard lipo sensor you will notice on the logical switch screen your total battery voltage is displayed in seconds (V2 tab). The individual cells read out correctly on the transmitter but not the total voltage. You must:
Turn on the transmitter. Connect the lipo sensor to the balance lead and connect the battery- all as before.
Go to the telemetry screen on the transmitter near the bottom it will list all your sensors.
They are usually in this order
#1. SWR
#2:RSSI
#3: RX battery voltage
#4 Cels  --- the total cell voltage of the battery
for example if you have a 4 cell battery the remaining lines will probably be:
#5  Cel1
#6 Cel2
#7 Cel3
#8 Cel4
highlight the line that says “Cels”-usually line 4. Long press enter and select “delete this sensor”.
Now highlight “discover new sensors” When one populates in line 4 you should be done. 
Go to the logical switches screen and the Cels should now readout in V not sec.
On your telemetry screen it will now readout the correct  total voltage.
You must do this for each model with the battery connected.
This cannot be done in open-tx companion

closed 2020-01-07 by kilrah