opentx/opentx #4307: opentx 2.2 RC9 taranis 9xd screen low intensity

benoit35690, 2017-01-27:
Taranis X9D
opentx 2.2 RC9
multiprotocol module (from Banggood) plugged
I have a message saying that the SD card version is not correct, but everything works (spash screen, sounds, pictures)

My screen is now in low intensity, I can barely read it.
The retro blue light is working properly.
Is there a way to check if the problem is hardware ?
Thanks

3djc, 2017-01-27:
A photo maybe ?

kilrah, 2017-01-27:
Check your contrast setting, but it's nearly certainly hardware. Heard about a similar fault a few times before.

benoit35690, 2017-01-28:
[image]
here is a photo
I didn't find any contrast setting. I do find a brightness setting, I play with it without success.
I will order a spare screen.
Could it be related to the presence of an external module ? A voltage regulator that would be overload or CEM that pollutes...
Thanks

projectkk2glider, 2017-01-28:
@benoit35690 you could play the elimination game to see if anything affects it:
 * remove external module
 * flash old firmware - that worked
 * etc...

There is a  contrast setting 
[image]

benoit35690, 2017-01-29:
I found the contrast and set it to 45. It is a little better.

I removed the multiprotocol module, flash back opentTx 2.2 RC8.
The screen is still at the same low intensity.
I flashed the RC9 again, and still same situation.
So it seems that my screen failure has nothing to do with the RC9 upgrade.
I will order a spare one and repair the Taranis.
Thank you for your help.

closed 2017-01-29 by kilrah