opentx/opentx #4310: 2.2RC9 Taranis lcd.getLastPos() issue

athertop, 2017-01-27:
in lua, if you display a number using RIGHT alignment then do a getLastPos() - this does not return the right hand position of the written number but the left. So here's some code to demo this:

lcd.drawNumber(61,10,getValue("Cmin")*100,MIDSIZE+PREC2+RIGHT)
xposCons=lcd.getLastPos()
lcd.drawText(xposCons,9,"c-",SMLSIZE)
lcd.drawText(xposCons,15,"min",SMLSIZE)

in RC8 this displayed the same as in 2.1 with the c- and min aligned after the displayed number, but in RC9 it displays c- and min on top of the number left aligned with the number

mpaperno, 2017-01-27:
I see the same thing in my Lua scripts here.  @3djc  I think it started happening after your updates to the fonts/display (since RC8).  Also the `MIDSIZE` period is now quite large  :)  (larger than `DBLSIZE`)

mpaperno, 2017-01-27:
Before:
[image]
After:
[image]

projectkk2glider, 2017-01-27:
Might be caused by fixes in #4157 and is probably intentional. @3djc will explain.

3djc, 2017-01-28:
There two different aspects to this.

YES, numbers and text drawing routine have been streamlined and will not produce the exact same results as RC8. There where bugs  and issues when unusual combination of attributes where used. All of that has been fixed, and the result is that you screens might look different, sometime by a pixel or two, but in several case, the difference can be significant (XXL+RIGHT for example).

The other one is lcd.getLastPos(). If you write RIGHT, the position you give to drawNumber is what you call your right hand position (61 in your example). It is logical that lcd.getLastPos() gives you the side you don't already know, the left one.

@Max, raise a good point about MIDSIZE dot, I'll have a look

mpaperno, 2017-01-28:
Are you saying `getLastPos()` returns different results based on how you aligned the last draw call?  Or it will now always return NOT the last position (the left/top-most one where it started drawing?) ?  😕  Seems like a major "feature" of getLastPos() was that it was consistent...

3djc, 2017-01-28:
Well, my understanding is that getLastPos() is there to give you an information you don't already have: where did your writing end, because depending on the fonts,.. a string pixel size will be different. If you align RIGHT, what is the leftmost position you reached ?

If you want to  do : ! speed : 500 you draw speed, then lcdgetpos and draw 500
If you want to do  : speed : 500 ! you draw 500 (right) then lcdgetpos and draw speed (right)

mpaperno, 2017-01-28:
I see your point about how it may be useful in some cases to know the leftmost position, but then it should be a new function, `getStartingDrawPos()` or whatever.  Changing behavior of an established API function that has been around for several years, and is undoubtedly used in hundreds of scripts, is a bad call IMHO.  And above all a function like that needs to return consistent results.

Imagine I just called a drawing function from inside run().  I don't know where it drew what it drew or how it was aligned, and don't care.  I need to know getLastPos(), period.

3djc, 2017-01-28:
Btw, it was not a purpose full decision or even known change, just the consequence of rewriting the code cleaner.

mpaperno, 2017-01-28:
Didn't realize that, sorry, thought it was intentional.  Can we change it back then?  :)

projectkk2glider, 2017-01-28:
Well technically if you are aligning RIGHT then the left most (first) character is the last one. 

@mpaperno are you sure that current behavior is **different** from the 2.1, because every 2.2 release prior to this fix was broken - the default alignment for certain LCD functions changed in 2.2 and this made a  mess we are trying to solve here. @3djc worked hard to fix it, but there might still be some fixed needed.

mpaperno, 2017-01-28:
Actually those screens are from 2.1, and right now I'm not sure if 2.2 RC8 had the same alignment problem, but I think it did.  I may have to install a fresh a fresh Windows RC8 to be sure (no LUA in Launchpad RC8 which works on my Mint).

This is from what I'm pretty sure is "stock" RC8:
[image]

I could install some older RC versions to test, if that would help.

3djc, 2017-01-28:
In 2.1, you could not drawText RIGHT, number where right aligned, text left. You could left align numbers but that wasn't quite right.

I think getLastPos() is very misleading or imprecise, but as Max mentioned, as been there long enough to stay (a bit more). I'm proposing:
- create two new functions, returning respectively left and right of last written string/number
- alias getLastPos() to the one returning right
- phase out getLastPos()

mpaperno, 2017-01-28:
That's true, the alignment attributes were a bit confusing/quirky in the older versions.  But I guess we learned to live with them.  :)

I like your proposal.  I think it's a good way to transition to a newer API, even though it's annoying keeping old stuff around.

athertop, 2017-01-28:
@3djc - totally agree with your suggestion. Will certainly help my scripts moving forward. Many thanks.

PR #4487 (2017-03-04): Fix lcdLastPos
Introduce lcdLastLeftPos to native code
Native lcdLastPos renamed to lcdLastRightPos

Introduce to lua:
lcd.getLastRightPos() as an alias to lcd.getLastPos()
lcd.getLastLeftPos()

Please note that on 128x64 ARM, it did require moving the code to X9/Horus style. Similar to what did happen on those platform, I had to do minor adjustment to test screens, but I do not anticipate any "omg my screen as changed !" (beside drawnumber alignement, see bellow)

The dot on MIDSIZE mentioned in #4310 as already been fixed in another PR.

Changes to alignment of lua drawnumber are NOT in the  [... cut]
closed 2017-03-04 by bsongis