opentx/opentx #4311: 2.2RC9 companion sim issue

athertop, 2017-01-27:
Reload LUA scripts button is gone, and pressing F4 (Mac) as suggested in the keymap popup, does not reload LUA scripts but instead displays the Telemetry simulator screen.
In fact, all those F keys on the Mac are screwed up displaying a different screen to that advised on the Keymap help panel.

mpaperno, 2017-01-27:
Partial duplicate of #4289, fixed by  efcfdf1 & 8c9f574f9271cc2623e6c4384611c396cfb56af3 .  For now please check the button tooltips to see the correct F keys to use (they haven't changed, just new ones added).  Thanks.

closed 2017-01-27 by mpaperno

athertop, 2017-01-27:
I was referring to the Tooltips in my OP above. It says F4 for 'Reload LUA', but pressing this does not reload LUA, but instead it opens the Telemetry Sim page. I can not find an F key that will reload LUA.

mpaperno, 2017-01-27:
> as suggested in the keymap popup

>as advised on the Keymap help panel

Are you sure we're talking about the same "tooltips?"  I mean when you mouse-over a button.  Though I realized that since the Lua button is missing, that's hard to mouse over it... :-/    

Also it seems I was wrong, and with the button hidden there's no way to force a Lua reload after all.  ☹️  Sorry about that.  I'm pretty sure there will be a new release out pretty soon though.