opentx/opentx #4312: RC9 361 on X12S beta - Backup Radio to File

MIKE-DAILEY, 2017-01-27:
RC9 361 on the X12S beta "Backup Radio to File" just displays progress bar with no progress.

projectkk2glider, 2017-01-29:
This was not yet implemented

MIKE-DAILEY, 2017-01-29:
Would it be a good idea to add it to the change log so folks know it is not implemented yet in Companion support for the X12S.

kilrah, 2017-01-29:
They'll see when they click it, a message box has been added.

MIKE-DAILEY, 2017-01-29:
Good idea!  Thanks.

raphaelcoeffic, 2017-12-16:
How is this feature actually different from the normal "read models"? Should it save the firmware used as well? Or anything else?

kilrah, 2017-12-16:
It does not go through the companion import/export layers so makes a true backup without the risk for a potential bug in companion to modify/corrupt the data.

raphaelcoeffic, 2017-12-16:
@kilrah ok, makes sense from a technical point of view, but as a user, I never really understood that feature. Considering what you've just explained, is there any point at all implementing that feature for the Horus? For my part, if I want to be sure, I just backup the relevant SD card directories.

kilrah, 2017-12-16:
No user knows what "the relevant directories" are... 
It's not a main feature but a convenient troubleshooting feature that's been used many times.

raphaelcoeffic, 2017-12-17:
@kilrah ok, thanks for the explanations. So basically, this is about copying the correct files (`RADIO/*` and `MODELS/*`) and zipping them together into an OTX file? Or some other format?

kilrah, 2017-12-17:
Exact

closed 2020-01-07 by kilrah