opentx/opentx #4313: User selection of USB mode

projectkk2glider, 2017-01-27:
With recent changes to USB code, the USB subsystem is again initialized just after the USB cable connection is detected. This means that USB mode could be one of the radio settings so user could without flashing different firmwares choose between serial, joystick and mass-storage.

The only obstacle could be the ST USB library that probably does not support different USB descriptors out of the box.

mpaperno, 2017-01-27:
Have you guys looked into a multi-device USB profile?  As in the device could act as HID or VCP or MSC at the same time.  I do have some experience with this on STM32 (F4, anyway) and it does work well.  The only downside is that it needs a custom .inf file for Windows, which would be unsigned (unless you guys have some connections).  Installing an unsigned .inf "driver" can mean a few hoops to jump through for Windows users (though still doable).  (The irony is that the actual driver .sys files are all stock from MS, it's just the inf file that is custom.) On Linux it involves some udev rules, just because the PID is different.

schwabe, 2017-01-27:
That might be feasible as optional feature but most people will probably better of without requiring installing custom infs and just disconnecting, changing a setting and connecting again.

kilrah, 2017-01-27:
I believe we had looked into mu.ti-device and at least on the Taranis' F2 we were limited to 2, so not worth the effort since a choice would still be needed as to which 2.

schwabe, 2017-09-03:
#5193 should close this

closed 2017-11-10 by kilrah