opentx/opentx #4314: 2.2 RC9: Accessing X7 thru companion possible?

dklueh79, 2017-01-27:
In the changelog of RC9 it's mentioned, that you could use companion again in order to write and get models, etc from the X7.
But: pressing the lower trims together during switching on, doesn't work any more. (White Screen)

What to do in oder to get a connection to companion? Same question for my horus

3djc, 2017-01-28:
This is not a support system, but a bug tracking one.If you have usage questions, please direct them to relevant forums, like http://openrcforums.com/forum/viewforum.php?f=45 or rcgroups.com

closed 2017-01-28 by 3djc

dklueh79, 2017-01-28:
Thanx, I know. It's a bug. Connecting a X7 isn't possible

dklueh79, 2017-01-28:
Thanx, I know. It's a bug. Connecting a X7 isn't possible . Flashing is possible, but during read out model data companion freezes

3djc, 2017-01-28:
Then you need to bring much more detailed evidence. (because it is possible)

dklueh79, 2017-01-28:
Fw  update: no problems.
Connecting X7 after that and trying to download models: the status hangs up during 0 %. After about 5 min windows 7 ( 10, too)  wants to close companion during no reaction. 
I didn't try Horus.

dklueh79, 2017-01-28:
Ah, no data within windows logs

kilrah, 2017-01-28:
Do you get something on the X7 screen' Do you see drives on the PC? can you access them?

dklueh79, 2017-01-28:
Yes, mass storage works.
But the X7 screen is blank white

dklueh79, 2017-01-28:
OK; reflashing twice- one without, the next again with massstorage. Works, But companion shows, no connection- in the background you coudl see, that it was transfered....

OS in Win 7 SP1 fully patched
[image]

projectkk2glider, 2017-01-28:
See #4318 for explanation.