opentx/opentx #4315: 2.2 RC9 Horus. Old model (2.1 eepe) converted LS to L1

BurtPR, 2017-01-27:
The conversion of a model from OpenTX 2.1 format to to 2.2 format for the Horus seems to have a minor glitch. I was using LS as input to a mixer. In the converted file it became L1 as input to the mixer.

BurtPR, 2017-01-27:
[EEPE etc.zip](https://github.com/opentx/opentx/files/736476/EEPE.etc.zip)

BurtPR, 2017-01-27:
eepe to make it obvious.

kilrah, 2017-01-28:
What source radio?

BurtPR, 2017-01-28:
Taranis Plus

projectkk2glider, 2017-01-29:
Similar to #4329 (conversion problems between Taranis 2.1 -> Horus 2.2 )

This is the text from there:
```
1. Launch Companion 2.1, type = Taranis
2. Create a model with potwarnings = MAN, all check boxes **dis**abled
3. Save file
4. Launch Companion 2.2, type = Horus
5. Open .eepe from step (3), note value of pot warnings:

Expected: all pot warnings disabled
Actual: LS and RS warnings enabled
```

bsongis, 2017-02-05:
If imported using a Taranis profile, the import is good, it means it's the Taranis => Horus conversion which is bad

commit b0eeeb4d (2017-02-05): Fixes #4315 (#4406)
closed 2017-02-05 by bsongis