opentx/opentx #4318: companion 2.2 RC9 Taranis 9xd error message

benoit35690, 2017-01-28:
Hello with companion 2.2 RC9
Menu "Read/Write"
"Read models and settings from radio"
the models and settings are properly read but an error message is displayed

[image]

fred99, 2017-01-28:
I have also this trouble on RC9 but not on personnal compilations based on Next source of yesterday and days before

projectkk2glider, 2017-01-28:
Should be fixed by #4291, please reopen if it still happens in RC10 (when it is released).

dklueh79, 2017-01-28:
Update: Only a few writes are successful! Reading  works always. But 7 of 10 write-tries failed!

projectkk2glider, 2017-01-28:
>But 7 of 10 write-tries failed!

Do you mean:
 * the radio actually did not get new changes
 * OR you got error in Companion only, but changes actually made it to the radio.

dklueh79, 2017-01-28:
The X7 doesn't get the data. Companion says 100% ok. Like my screenshot

projectkk2glider, 2017-01-28:
Our test show that it is working (at least now close to RC10), so wait until RC10 is out and reopen if issue remains.

closed 2017-01-28 by projectkk2glider