opentx/opentx #4321: OpenTx 2.2 RC9 X9D multiprotocol

benoit35690, 2017-01-28:
Hello
Taranis X9D
OpenTX 2.2 RC9
multiprotocol module from Banggood
http://www.banggood.com/CC2500-NRF24L01-A7105-CYRF693-4-In-1-RF-Module-With-Case-For-Futaba-JR-Frsky-Transmitter-p-1116892.html
no modification, this module is said to work in serial out of the box.

Spektrum bind OK in PPM mode
https://youtu.be/A7lqG5h1Iu0

Skektrum bind KO in serial mode
https://youtu.be/SXX7ngi66Ts

You can see also on these video my screen issue. I will order a spare screen.
This is the first time i use the module, sorry if I am not using it properly.

[DSM.zip](https://github.com/opentx/opentx/files/736969/DSM.zip)

3djc, 2017-01-28:
There is absolutely no way with Bangood to know what firmware is delivered with it,  so unless you have a proper flash and debug done,  there is not much more we can do

schwabe, 2017-01-28:
Spektrum serial options changed. THe firmware from banggood is seriously outdated. We don't support the old Banggood firmware anymore. (You can give the right options with custom proto)

closed 2017-01-28 by schwabe