opentx/opentx #4322: More Trimmers

th9xer, 2017-01-28:
Hi,
I would like to replace some of my three position switches with spring center switches and have the ability to use them like trimmers for altering global variables and use them as a source in inputs and mixers. Just like overloading the existing trimmers. 

I have tried to do this task via functions and logical switches and it almost works but there are several problems:

.I can get repeat while holding the switch in using the "TIMER" function but only at 5 clicks per second. This is too slow.
.There is no click rate acceleration the longer you hold the switch in.
.There is no trim step adjustment.
.There are no audio queues as you get with a trimmer (I can use functions and logical switches to get a center beep but not the "how far from center" beeps you get with a trim).

Maybe a special function, or better as part of the hardware settings ?

bsongis, 2017-02-25:
What would be the best solution to do this, an AdjustGVAR "special" source ?

th9xer, 2017-02-25:
It would be nice to also use the new trimmer as a source for mixers, just like you can with the existing overloaded trimmers. So maybe two special functions ?

closed 2019-11-15 by stale[bot]