opentx/opentx #4324: dont allow handset to power off if receiver is still connected.

geofrancis, 2017-01-28:
if you go to turn off the handset and its still connected to a receiver it should prompt for a confirmation and warn that a receiver is still connected before shutting down.

mpjf01, 2017-01-28:
A sound added safety feature. I'm not sure that the five second delay is needed though, a prompt that needs to be responded to before the radio turns off ought to be enough. There are legitimate reasons for switching off with Rx still powered such as testing FS and that should not be made too difficult.

kilrah, 2017-01-29:
Should be optional with a radio setting.

geofrancis, 2017-01-29:
Make it optional but should be on by default.

tubeme, 2017-02-13:
I think this should be well thought. Definitely optional. I might want to fly the take off and landing of my mission with the Horus but then once in the mission I might want to safe battery power of the RC and switch it off. The battery power of Horus seems like the one on X9E and we are in a big trouble with this bat pack inside.

Another issue here is how do we shut down the RC if opentx hangs and we have only soft button for power down? How do we reset the Horus in this situation?

kilrah, 2017-02-13:
> Another issue here is how do we shut down the RC if opentx hangs and we have only soft button for power down? How do we reset the Horus in this situation?

Press for >10secs and release.

geofrancis, 2017-02-13:
The percentage of people running auto missions and turning of the handset will be less than 1% so having it on would help the majority of people by making sure models are not left on.

mpaperno, 2017-02-13:
The point of this would be what? Prevent accidental use of the power button?  Remind the pilot they're still flying?  Remind the pilot to shut down the model first?

Are any of these actual problems for people now?  Just asking...

Personally I'm not a big fan of soft-power switches.  Maybe I'm just biased because they never worked right for years and years, often have to go pull the battery in the end (or find that paper clip to reach the tiny little recessed Reset button).

I'm also wary of imposing changes on people in the interest of "safety."  We have quite a bit of that in the US...  ;-)

Most definitely optional please, perhaps compile-time option even (esp. if it's going to affect power circuit handling in the hardware), disabled by default.

geofrancis, 2017-02-13:
Can you honestly say you have never left an aircraft on after switching of your handset?

mpaperno, 2017-02-13:
Only on purpose.  Model goes off first, then radio (opposite of startup procedure).  Always.

kilrah, 2017-02-14:
> Can you honestly say you have never left an aircraft on after switching of your handset?

Yes, even intentionally but it's not a problem because I make sure to configure my failsafes properly.

geofrancis, 2017-02-14:
I heard that argument from a uav sales company, they disabled battery warnings, low battery failsafe, rc failsafe, everything apart from the main flight battery voltage on the handset. 

According to them it was up to the pilot not to get into trouble and none of the failsafes were needed because they had the voltage on the screen in front of them.

Surprise surprise the machine crashed 3 times, 3 times basic alarms or warnings would have saved it.

Not everyone has been doing this for years. I run multi rotor clubs so I see the beginners and all the mistakes they make.

What about just a beep before the handset switches off if rx is connected? So there is some indication that rx is connected without delaying shutdown?

mpaperno, 2017-02-14:
That's a bit of a stretch between a low battery warning and the radio deciding when you're allowed to turn it off.  IMHO.

At the club I used to go to "back in the day" there was a simple rule that if you couldn't follow basic safety procedures after they were explained to you, you couldn't fly there.  Times change I guess.

> What about just a beep before the handset switches off if rx is connected?

That I like.  Perhaps even a special function that could be executed on shutdown if RSSI > 0 (or something like that).  Not sure if that's possible now w/out re-engineering the whole shutdown procedure, but perhaps.

geofrancis, 2017-02-14:
Hmm that could be interesting, if you could use the handset shutting down as an input to trigger a special function . Then you could get it to do whatever you want before shutdown?

bsongis, 2017-02-14:
I will take this one once 2.2 is released

3djc, 2017-07-21:
Proposing a way of operation;

Radio with bistable on/off (x9d):
1) Off switch 
2) Audio alarm + warning screen
3) model is turned off : immediate shutdown
4) ignore warning pushed : immediate shutdown

Radio with temporary POWER (Horus,x7, x9d,..)
1) start pushing POWER
2) audio alarm + display alarm
3) end of power off delay: warning screen with exit or enter
4) model turned off : off
    exit : back to normal operation
    enter  : off

geofrancis, 2017-07-22:
Sounds good

mpjf01, 2017-07-23:
So is it proposed that this still be optional - if so a compile time option or a SF or ?

geofrancis, 2017-07-24:
why dont you put it in the handset menu as an option.

3djc, 2017-07-24:
If it gets in, it will be optional as some have voiced that  they do not want it. Currently, it's been prototyped as a radio setting

geofrancis, 2017-07-25:
expanding this to changing models as well as powering off is a good idea.

Bravedan, 2017-07-25:
If you change model leaving old model on, how it this going to be dangerous as with different Rx numbers the "old" model is incapable of running.  I'm confused!?

schwabe, 2017-07-25:
@Bravedan  RX goes to failsafe which may have throttle on a non zero position.

Bravedan, 2017-07-26:
Ah, right, thanks.  Personally I always set zero throttle so would not have that situation.

3djc, 2017-08-14:
Dealt with in #5126

closed 2017-08-14 by 3djc

LapinFou, 2017-09-15:
So nice enhancement!!! I nearly lost a plane by inadvertently switching off my X9D+ (despite the difference I confused the power switch with the elevator trim). I think it is a very good security feature.
Well done. 👍