opentx/opentx #4326: N361 - Horus unstable after transferring data from Companion

RC-SOAR, 2017-01-28:
Horus, OTX & OTX Companion 2.2 N361

Steps

- Connect Tx to PC in mass storage mode (tx on)
- Create or open a model in Companion
- "Write models and settings to radio"

The screen goes blank. Tx still responds to key presses (throttle, switch warnings). No option but to switch off. Same after powering off and powering on, responds to key presses but can't see anything.

Simple or complex setup, no difference. 
Deleting the Models and Radio folders (via card reader) restores normal operation. 
If I just set a model up on the transmitter then all is fine.

kilrah, 2017-01-28:
Backlight is simply off, you can see the screen worls if you shine a light at it. That shouldn't happen...

projectkk2glider, 2017-01-28:
I can trace this, remember I have hw serial :smile:

kilrah, 2017-01-28:
No need, jsut missing default in companion and some backlight driver value in prod version

projectkk2glider, 2017-01-28:
Cool, I tested reading existing setup and writing back with latest `next` and everything was OK (on first glance).

tubeme, 2017-01-28:
I must add once I write the Presets from Companion to the Horus, everything gets very messy in the OpenTX. Once it changed the default color. The other time it just renamed the Presets. All weird things happen.

I even tried the following scenario. I reset the Horus with deleting the folders mentioned aboveby @RC-SOAR . Then I created 3 models in 3 different categories in the Horus. Then I read the presets with the Companion. Then I just write them back, nothing changed. Again Horus is messed up heavily.

closed 2017-01-28 by kilrah

RC-SOAR, 2017-01-28:
Wow that was quick - thank you!

commit 32dc0dd2 (2017-01-28): [companion] Missing default for backlight off on Horus (fix #4326)