opentx/opentx #4329: N361 Companion:  Pot warnings incorrectly translated

RC-SOAR, 2017-01-28:
1. Launch Companion 2.1, type = Taranis
2. Create a model with potwarnings = MAN, all check boxes **dis**abled
3. Save file
4. Launch Companion 2.2, type = Horus
5. Open .eepe from step (3), note value of pot warnings:

Expected: all pot warnings disabled
Actual: LS and RS warnings enabled

projectkk2glider, 2017-01-28:
Similar problem on `next`, every time models are written to radio `switch warnings` are modified. The problem is two fold:
 * actual `radioData` is modified by this action, this is caused by this line/construct (last line)
```
bool OpenTxEepromInterface::saveToByteArray(const T & src, QByteArray & data, uint8_t version)
{
  if (version == 0) {
    version = getLastDataVersion(getBoard());
  }
  QByteArray raw;
  M manager((T&)src, board, version, 0);
```

Problem is that new instance `manager` modifies `src` data. In my opinion it should work on a copy of it. This is the constuctor for `manager`:
`OpenTxModelData::OpenTxModelData(ModelData & modelData, Board::Type board, unsigned int version, unsigned int variant):`

 * the switches warning is modified here (why is this conversion needed?) https://github.com/opentx/opentx/commit/125c301c5603207f3fdef6b9688f9297075520cc#diff-5c7b48e246a21f92cddceccac9857fbaR3448

projectkk2glider, 2017-01-29:
My issue should be fixed by #4331.

The original issue of conversion between Taranis 2.1 -> Horus 2.2 remains.

projectkk2glider, 2017-01-29:
Closing this, report transfered to https://github.com/opentx/opentx/issues/4315#issuecomment-275899641 which looks like a similar issue.

closed 2017-01-29 by projectkk2glider