opentx/opentx #4332: RC10: Multiprotocol status is not displayed in X9D simulator

dev-fred, 2017-01-29:
If I compile NEXT under MSVC, I got all informations under  External RF when Multiprotocol is selected:

[image]

This is not the case with RC10 and RC9:

[image]
:

kilrah, 2017-01-29:
You compiled without enabling multimodule.

closed 2017-01-29 by kilrah

dev-fred, 2017-01-29:
Sorry, but I am not sure to understand: 
I have compiled with -DTRANSLATIONS=FR -DHELI=NO -DGVARS=YES -DLUA=YES -DMIXERS_MONITOR=YES -DTIMERS=2 -DMULTIMODULE=YES -DHAPTIC=YES
and that works: 1st screenshot.
The simulator uses opentx-x9d-simulator.dll  which is installed by companion-windows-2.2.0N362, and that doesn't work correctly-> 2nd screenshot.

kilrah, 2017-01-29:
That's because the simulators that we currently distribute as nightlies don't have the MULTIMODULE option enabled.

dev-fred, 2017-01-29:
Ok," You compiled" is not me ; )

kilrah, 2017-01-29:
Yep, given you mentioned you compiled next I understood "with RC9 and RC10" as you compiling those tags instead of next.

3djc, 2017-01-29:
And I'm unsure what would be the value of having it in the simulator. We need to have a middlepoint choice for simu

schwabe, 2017-01-29:
well the simulator does not simulate the multimodule telemetry so it is indeed missing. We could replace that string with something that says "Multi module not emulated in simulator"  (too long but you get the idea) but it is a status only.

dev-fred, 2017-01-29:
All other fieds, like in Companion, are accurate, it's a very good job:
[image]

One days, when we will have options for the simu like the firmware, everybody will be happy, specially with local language.

kilrah, 2017-01-29:
That will be solved the day simulators are compiled and downloaded on demand, and will thus match your profile settigns like the radio.