opentx/opentx #4334: Taranis X9D+: UAV image not loaded on very first start

RipperDrone, 2017-01-29:
Switching on the Taranis X9D+, I am getting the opentx splash screen, and then get a small openTX logo as a UAV image in place of the chosen copter image. When selecting another model and then re-selecting the former UAV, the image shows ok. Seems a problem of first boot/image loading...

projectkk2glider, 2017-01-29:
Which OpenTX version?

RipperDrone, 2017-01-29:
2.2 RC10. Sorry, will modify the title accordingly.

ryphon, 2017-02-02:
I found this as well, however it only takes a couple seconds for me until it pops up with the correct photo.

RipperDrone, 2017-02-02:
Funny, mine stays wrong image. I have to explicitly select a model manually to get the correct image.

ryphon, 2017-02-02:
I notice that once it loads correctly, it never fails again. It only displays the wrong image on startup.

RipperDrone, 2017-02-02:
Correct, but every time you hard switch the Taranis off and back on again, same error occurs.

bsongis, 2017-02-14:
SD was initialized AFTER model load, which is bad

commit 27b12555 (2017-02-15): [X9D/X9D+/X9E] Fixes #4334 (#4451)
closed 2017-02-15 by kilrah