opentx/opentx #4337: RC9-X9D+: Multi status line stops displaying new status when protocol change

powercycle, 2017-01-29:
Setup: radiio has multimodule 1.1.6.5 with Q303 as a compiled protocol (sub CX10WD)

How to see problem:  
Set protocol to Q303, status shows module version.
Cycle through protocols until No Multi_Telemetry is found.
Cycle back to Q303, No Multi_Telemetry still displayed.

How I fix it:
Power cycle the radio on the Q303 as the selected protocol.

projectkk2glider, 2017-01-30:
Please verify this issue in RC10.

schwabe, 2017-01-30:
Also what is the status of module. Your report sounds as if the code of the multimodule is crashing. NO telemetry is displayed if OpenTX hasn't received telemetry for 2s.

powercycle, 2017-02-06:
Using 1.1.6.5 and the red light says the module is OK.  I'll try to load RC10 to see if this has resolved.

powercycle, 2017-02-06:
RC10 Doesn't change anything...I thought I saw something on RCgroups that talked about this issue, so it could be a module firmware issue.

powercycle, 2017-02-06:
OK.  I was able to get to it sooner than I thought.
It happens as soon as I get to an entry that says no Multi_Telemetry.
However, if I move through to a protocol that has telemetry, power cycle the X9D+, it works as expected until I reach no Multi_Telemetry again.

3djc, 2017-02-06:
I think you can power cycle even with the same protocol. What you see is the module not sending data when it should, module issue imho

powercycle, 2017-02-06:
Only indicator I have of that is the Red LED stays on solid (except when going through bind).   Interestingly, Invalid protocol does make the LED blink like it should.

pascallanger, 2017-02-06:
So basically multi works perfectly fine but you don't have the status displayed on the radio.
@powercycle, Can you give the list of all protocols you have compiled?

I've tested again on ersky9x to switch between the 16 protocols I have loaded on my module (99% of Flash with 32462Bytes). I can't reproduce this behavior by either going slowly or fast, or select PPM then go back to Multi or... I've also tested with and without INVERT_TELEMETRY, both works.

This seems to impact only OpenTX using the dedicated MULTI_TELEMETRY feature (written by schwabe).
Can someone from the OpenTX dev team would be able to try to reproduce the issue and dump the telemetry pin signal?
Pascal

3djc, 2017-02-06:
We have not been able to reproduce so far

pascallanger, 2017-02-06:
@powercycle:
Can you give which exact version of OpenTX you are using (with compilation option)? or the binary used to flash the radio?
Can you send your _config.h?
With these files someone from the OpenTX team should be able to reproduce the problem.
Pascal

powercycle, 2017-02-07:
Don't know the compilation options because came from RC10 deb package 32-bit
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/756364/_Config.h.txt

schwabe, 2017-02-07:
from a quick look from your config, at least this is wrong:

```
/**************************/
/*** RF CHIPS INSTALLED ***/
/**************************/
//There are 4 RF components supported. If one of them is not installed you must comment it using "//".
//If a chip is not installed all associated protocols are disabled.
//4-in-1 modules have all RF chips installed
//!!!If a RF chip is present it MUST be marked as installed!!! or weird things will happen you have been warned.
#define A7105_INSTALLED
#define CYRF6936_INSTALLED
//#define CC2500_INSTALLED
#define NRF24L01_INSTALLED
```

kilrah, 2017-02-07:
> Don't know the compilation options because came from RC10 xenial deb package 32-bit

He means what firmware options you've selected in companion before downloading/flashing firmware.

pascallanger, 2017-02-07:
Ok I can reproduce the issue. Like schwabe mentionned the problem is due to the commented line "//#define CC2500_INSTALLED". The problem is basically that the CC2500 is not supposed to be there and therefore not configured so he is interferring on the SPI bus. If you only write to the spi bus it is fine but as soon as you try to read then things become really messy. For example the Esky protocol hangs since the nrf24l01 fails to tell that the packet has been sent simply because the cc2500 is talking at the same time...

I don't know how to get people to configure their modules correctly. By default all modules are enabled. I've added explanations and warnings with exclamation marks and capitals already: "//!!!If a RF chip is present it MUST be marked as installed!!! or weird things will happen you have been warned."

3djc, 2017-02-07:
Yeah, we have similar issue with sd card content. No matter how you explain, there seem to be folks that think you didn't _really_ mean that, and ignore it ! Closing the issue as it is not OpenTX (or your software for that matter ! Thanks for your help Pascal

closed 2017-02-07 by 3djc

schwabe, 2017-02-07:
@pascallanger as the INSTALLED code is small and cs to modules that are not installed should not hurt, maybe just remove those defines or at least the defines around the CS handling?

pascallanger, 2017-02-07:
This is not for flash space constraint that I've introduced it and this is how it was done some time ago. The change has been introduced for I think BikeMike because he was compiling multi for goebish's module which has only a NRF24L01... I was looking into it while I've received your email. I'm going to change the #ifdef to not look for the module presence but if the pins for that module have been declared. So in case you want to compile for another board which does not have a module or another, you just remove the pin declaration and you are done. So yes I'm working on it :-)

powercycle, 2017-02-08:
Wow, that's embarassing.   Thanks for figuring it out.