opentx/opentx #4340: Random changes of offsets and switch positions Horus

marcogpw, 2017-01-30:
Have observed that writing a models file from PC based computer running OpenTX 2.2.0 RC9 and RC10 to the Horus (on RC10 as well) randomly changed the inputs offset settings ranging from 2 to 49 %. Also the positions of the switch warning panel changed randomly, resulting in lots of switch warnings when powering on the Horus adn when switching models. Don't know yet if these changes happen at transfer to the Horus, during saving or maybe writing it twice in a row, since there is no transfer progress monitor for writing the models file to the transmitter.

projectkk2glider, 2017-01-30:
This should be fixed in R10 by https://github.com/opentx/opentx/pull/4331. Are you sure it sill happens in RC10?

projectkk2glider, 2017-01-30:
I can't reproduce with RC10.

marcogpw, 2017-01-30:
I will get my transmitter and test with RC10, standby for updates, thank you for looking into it though !

marcogpw, 2017-01-30:
You are completely correct, no more random offsets and switch positions anymore. I tried everything, reading/writing and doing things you would normally not do:reading from Horus and writing 4 time in a row ... sorry for the complaining

closed 2017-01-30 by kilrah