opentx/opentx #4341: Poor display of Input and Mixer lines

projectkk2glider, 2017-01-30:
Using Qt 5.2.1 on Linux the display of Inputs and Mixer lines is truncated on the bottom (notice missing bottom part of `g` character).
[image]

I will verfy later if the same problem exist with Qt 5.7.

3djc, 2017-01-30:
Doesn't seem to exist on Debian Jessie compile (qt5.3)
[image]

projectkk2glider, 2017-01-30:
Nice, this means it will be difficult to solve...

projectkk2glider, 2017-01-30:
Using Qt5.7 on the same base system is good( but notice different window colors because of non-native Qt installation)
[image]

I also do not understand why icons on the buttons are **different** I am using the same theme - Yerico.

mpaperno, 2017-01-30:
https://wiki.archlinux.org/index.php/Uniform_look_for_Qt_and_GTK_applications

http://askubuntu.com/a/775519

mpaperno, 2017-05-12:
Found this Qt5 "configuration tool" for setting/editing Qt app styles. 
http://www.webupd8.org/2015/11/configure-qt5-application-style-icons.html

Better to install from:
https://launchpad.net/~hda-me/+archive/ubuntu/qt5ct/+packages
or http://ppa.launchpad.net/nilarimogard/webupd8/ubuntu/pool/main/q/qt5ct/

Or an easy build from source: https://sourceforge.net/projects/qt5ct/files/?source=typ_redirect

Qt5.7 does not support GTK3 themes (but 5.8 supposedly does). However the tool can also be used to configure a completely custom style for the default Qt Fusion theme, so any/all Qt apps can be affected at once.

> I also do not understand why icons on the buttons are different I am using the same theme - Yerico.

I still don't get that either, but I see the same thing.  Some icons or actions/shortcut icons are being matched/replaced by the GTK style, but only for some themes (eg. with monoblue they don't get replaced).  I'm sure there are some heuristics involved...

PR #4919 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes:
  * Fix/improve margin/padding issues, consistent between inp/mix views, better handling of x-platform and older Qt (closes #4599, closes #4341);
  * Allow dropping onto lines instead of only between them;
  * Fix dropped data always going to line above the actual drop target (closes #4622);
  * Force DnD _move_ operations between windows to _copy_ data instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
  * Add alternate-row coloring for each Input/Mixer gr [... cut]
commit 3f5a3995 (2017-06-10): [Companion] Inputs and Mixes lists formatting & DnD fixes: (#4919)

* Fix/improve margin/padding issues, consistent between inp/mix views, better x-platform and older Qt (closes #4599, closes #4341);
    * Allow dropping onto lines instead of only between them;
    * Fix dropped data always going to line above the actual drop target (closes #4622);
    * Always force DnD move operations between windows to copy instead (fixes deletion of any selected line in target window, and source line was never actually removed but looked like it was);
    * Add alternate-row coloring for each Input/Mixer g [... cut]
closed 2017-06-10 by bsongis