opentx/opentx #4343: Switch X9E

hobby1946, 2017-01-30:
Switch > SH (2-Position and Toggle) do not move back to -100% (0%)

hobby1946, 2017-01-30:
in Companion N362 Simulator !

projectkk2glider, 2017-01-30:
I actually can't reproduce this with N362! Using Mixer with source SH on CH1:
[image]
[image]

hobby1946, 2017-01-30:
Switch SH is OK !

Switches  SI -> SP is not ok !
Start ->           -100%
Switch on ->     100%
Switch Off -       0%

closed 2017-01-31 by mpaperno

mpaperno, 2017-01-31:
It finally dawned on me why my fix actually did anything... 😄    The firmware expects a "3-position signal" for switch slots that can take them.  So that's why the 2-pos. switches worked correctly in simulator when in standard 2-pos slots (except Horus SH), and not when installed in a 3-pos slot (same with `simu`). 

Check the outputs at channels 5 - 8  (upcoming next-gen UI with separate outputs panel... WIP  😉  )

[image]