opentx/opentx #4344: Language Radio Settings

hobby1946, 2017-01-30:
The mask for radio settings in Companion N362 is in English at the German language version.

schwabe, 2017-01-30:
Yes. OpenTX is a community project and so far no German has done the work to translate everything in Companion. If you are interested in doing this, join our chat to get you started.

hobby1946, 2017-01-30:
that is not the problem

In opentx n359 was the mask ok (German) in German Version
see:  https://workupload.com/file/SgKrVCb

In opentx n362 is the mask in english in the German Version ...
see: https://workupload.com/file/Dw5xaup

That would have to be corrected easily

projectkk2glider, 2017-01-30:
There were so many changes to the code that many existing translations got obsolete. See https://github.com/opentx/opentx/blob/next/companion/src/translations/companion_de.ts Everywhere you see `<translation></translation>` is a missing translation. 

We welcome anyone to help with translations.

hobby1946, 2017-01-30:
In opentx n359 was the mask ok (German) in German Version

Why not now in n362 ???

Sorry for no help by translation.
I am 70 years old and my english is not so good!

schwabe, 2017-01-30:
To translate @projectkk2glider again in German:

Es gab so viele Änderungen zwischen den Versionen, dass die alte Übersetzung veraltet ist und an die neuen Änderung angepasst werden muss.

hobby1946, 2017-01-30:
OK,
hier sind doch nur ein paar Worte zu ändern .

https://workupload.com/file/Dw5xaup

Radio Settings in Sender Grundeinstellungen
Simulate TX in Sendersimulation
Size in Größe

wo ist der Quelltext dafür ?

schwabe, 2017-01-30:
Wie @projectkk2glider sagte: https://github.com/opentx/opentx/blob/next/companion/src/translations/companion_de.ts

kilrah, 2017-01-30:
Es gibt eigentlich Hunderte von Sätzen die mussen wieder übersetzt werden. Mit QT Linguist kann jeder der es machen will alles sehen. 
Vielliecht @hrenz ?

closed 2017-01-30 by kilrah