opentx/opentx #4346: N362 Companion, Simulator works only one time, no reaction after changing somthing at the mixer

hrenz, 2017-01-30:
Hy,  here my step by step to generate the error, and the solution
open Companion
load a ready modell   (QX7, X9E X9D)
open the Simulator at Companion
Simulate it , works, ok, fine.
Close simulator,
Change things at inputs or mixers
reopen the simulator
try to simulate, but no reaction,  no errors, only no reaction at simulation
Close Simulator
---- 
ok, then I save my changed  model
reopen the simulator
Simulate, works again, ok, fine
------------------------------
After each changings, i have first to save the file
then I can open the simulator and it works.
independend form the TX profil  ( X9D, X9E, QX7)
same at  N362 and N361

hrenz, 2017-01-30:
Hy,
my System is Window7 Prof.
after start Companion, open modell, start simulator,
no reaction at Simulator, 
i have to Reload Radio
the simualtor works,  reaction at the channel sliders, ok.
Reload Radio again, then no reaction at simulator,
simulator window size changed
reload Radio again, 
Simulato works  again
Sometime it work, somtimes not

projectkk2glider, 2017-01-30:
This is most likely fixed by #4342

closed 2017-01-30 by projectkk2glider