opentx/opentx #4347: [OTX 2.2]Missing French translation

LapinFou, 2017-01-30:
Hi folks,
Some french translation are missing:
http://frskytaranis.forumactif.org/t4742p25-opentx-rc9-n361#54226

The 6 positions pot is also not working correctly. Sx1, Sx2, ... Sx6 are not usable in the LS.

projectkk2glider, 2017-01-30:
There were so many changes to the code that many existing translations got obsolete. See https://github.com/opentx/opentx/blob/next/companion/src/translations/companion_fr.ts Everywhere thay you see `<translation></translation>` means missing translation. 

We welcome anyone to help with translations.

3djc, 2017-01-31:
Je peux te donner les infos pour aider a traduire si tu veux, je n'ai pas vraiment le temps pour le faire en ce moment :s

LapinFou, 2017-01-31:
@projectkk2glider 
I will try to do the translation and push the file.

@3djc 
Merci 3djc. Je pense que je vais pouvoir me débrouiller. J'ai déjà utilisé git. 😄 
Je ne fais pas de prog, maris ça, c'est dans mes cordes. Je vais essayer de faire cela d'ici la semaine prochaine.
Par contre pour le problème de pot à 6 positions, faut-il que j'ouvre un autre ticket ? Ou tu as vu et constaté le problème sous Companion ?

3djc, 2017-01-31:
I started here, to be edited with Qt linguist : https://github.com/opentx/opentx/blob/3djc/comp-french-uncomplete/companion/src/translations/companion_fr.ts

Yes, we know that 6POS may not work

LapinFou, 2017-02-01:
I cloned the project today. I started to do the translation with Qt Linguist (great tool BTW). I already spent 2 hours and it is not yet finished. I hope, I will able to do my first commit tomorrow. 😄

LapinFou, 2017-02-02:
OK. I've finished the French translation. I pushed the modified file in your branch "3djc/comp-french-uncomplete"
Next step is to do a "pull request", but I don't know how to do it.
My commit number is here: https://github.com/opentx/opentx/commit/e911922430dff23dbe1a39a68cea748855889b78

LapinFou, 2017-02-02:
I think (hope) that the pull request is OK: https://github.com/opentx/opentx/pull/4376

closed 2017-02-03 by 3djc