opentx/opentx #4348: trainer port futaba

janlebelg, 2017-01-30:
it does not work and when putting in a radio T4YF in the futaba connector you loos the sd cart
unplug radio and restart and everything back ok
v361 horus

projectkk2glider, 2017-01-30:
I think it is a hardware issue, there were some talk about this somewhere (I forget where).

schwabe, 2017-01-30:
IIrc tandango in the general channel. IIrc only trainer in or trainer out is connected and it appear only fixable by modifying hw.

kilrah, 2017-01-30:
Futaba trainer port is output only.
SD card loss when plugging something while the radio is on was mentioned before, probably upsetting the power rail to the SD card. Solution is to plug/unplug only when the radio is off.

That's all hardware and out of our control.

closed 2017-01-30 by kilrah