opentx/opentx #4349: n362 Companion: shortcut Alt-S malfunction

strgaltdel, 2017-01-30:
Hello,

shortcut "ALT-S" seems to have some problems even in n362 / Windows.
Directly after starting companion, open an eepe (or .otx) and highlighting a model, ALT-S is without functionality.

Tested with Win10 & Vista, Horus and X9e Profile.

After some editing and simulating by right mouseclick, ALT-S starts operating.
Unfortunately, couldn't find any systematic "root cause" which brings this shortcut into live.

btw
"RELOAD LUA" not suported in horus Sim but in X9e?

regards
Udo

projectkk2glider, 2017-01-30:
The current code only enables `Alt+s` shortcut when model is selected. This should be changed, since simulation can be run even without any model selected or even present.

strgaltdel, 2017-01-31:
Just for clarification:
even if a model is selected, Alt-S isn't enabled.
[image]
.. in this situation shortcut is without action.

As well i'm editing this model it wouldn't work directly.

Some minutes later, after some editing and simulation, it starts to run.....

kilrah, 2017-01-31:
The print/cut/copy/paste buttons also stay greyed until after the window has lost/gained focus a few times.

PR #4452 (2017-02-15): Menu updates handling cat/models for Horus
Fixes Alt-S not always activated when needed
Fixes edit menu for Horus (activated for models, not category)

This fixes #4349
closed 2017-02-15 by bsongis