opentx/opentx #4353: RC10 N362 - User Interface Color Enhancement

MIKE-DAILEY, 2017-01-30:
The User Interface Color Enhancement was discussed many months ago.  Are you still planning sometime in the future to change the "User Interface" "Main Color" options we can change the secondary color that is automatically calculated now from the main color?  Right now as I adjust the main color near what I want the secondary becomes something I do not want.

kilrah, 2017-01-30:
There's currently a bug with the automatically calculated colors.

Beyond that, themes were created so they could be modified or created by users, so we likely won't spend extra time on such details.

projectkk2glider, 2017-02-01:
Before my commit:
[image]

After commit, default colors and some variants that now work properly:
[image]
[image]
[image]

closed 2017-02-01 by projectkk2glider

bsongis, 2017-02-01:
Yes really better, thanks!