opentx/opentx #4356: Suggestion - Splash screen warning for use of -ve values on Inputs screen (trim reversal issue)

Matty2B, 2017-01-31:
I have helped a lot of newcomers to OpenTX in the past few years, and the #1 "gotcha" I've noticed is an unexplained reversal of trims. When I dig in this is always caused by the use of negative values on the Inputs screen to reverse a servo. 

I am not suggesting -ve values are removed as an option - clearly there could be situations where they are the right thing to use, particularly if the source is MAX. Instead could we have a splash screen warning displayed the first time users try to input a negative value on the inputs screen for a given model? I think this would really help beginners to OpenTX. Many thanks!

Matty2B, 2017-03-06:
Any update/decision on this one? Many thanks.

schwabe, 2017-03-06:
Well the problem is where do you want to start and where do you want to end with this warnings?
If you implement a few of the easier trap as warning, you feel give users a false sense of safety since they might expect that they are warned if they do anything stupid. Also if users think that is a good idea to reverse servo in the inputs and not in the mixers or the outputs that already have a reversing channel checkbox, they probably have not understood the inputs, mixers, outputs system.

Also users might be annoyed by the warning, especially by something as simple as this.

In my personal preference is not to add these types of screens, for reason alone that this will become quickly an unmaintainable mess.

3djc, 2017-03-06:
Same here : more warnings really means less time spent by users reading (let alone thinking) about those. We should really keep those for those things that matters.

mpaperno, 2017-03-06:
@Matty2B By "splash screen" you do NOT mean at radio startup, like the throttle/switch warnings, right?  Just a warning when actually editing the input? 

The "first time" would pretty much need to be "first time after every restart" of the radio... I don't think there's a mechanism to flag "warnings already shown" per model or radio (this would require storing a Boolean value per warning somewhere permanent), and I"m not sure this would warrant adding one.

I agree that warnings and prompts in general need to be approached with a critical eye lest they get too numerous and onerous (popup warnings esp. are often more annoying than useful).  OTOH a "self-documenting" GUI is a good GUI.   Perhaps some kind of icon or other non-obtrusive warning indicator could be used instead?  Something like that could be useful elsewhere as well.  Of course every pixels counts on the smaller screens to it may not be practical.

Matty2B, 2017-03-08:
I agree self documenting GUIs are best, but on this particular issue the interface is not telling people visually in any way that using -ve values on the input screen for primary controls will give them issues. In my head it was a pop up the first time you tried to set a -ve value after startup, bit if that isn't possible any other type of non-invasive visual cue (icon or check mark) would be helpful; I have literally seen this

Matty2B, 2017-03-08:
If a pop up the first time you adjust an input to a -ve value is untenable/too annoying then a check mark or icon would be fine too; anything to give a visual indication that -ve values on inputs for primary controls are a bad idea. I have had to help beginners to OpenTX with this issue online and in person 30-40 times at least.

closed 2019-11-15 by stale[bot]