opentx/opentx #4359: RC10 N362: Not the correct active model on writing models in Horus.

tubeme, 2017-01-31:
So I just got my Horus and I'm trying all the different aspects of OpenTX on Horus so I found a minor bug.

When I have all the models saved (read from Horus) then the Active profile becomes in bold letters in the OpenTX.

But then when I transfer back the presets (write to the Horus)  on start up the Horus just chooses the last preset (I see the number indexing of my presets) as an active one. So I have to go back and choose back my previous (saved) active profile.

I don't know if I explained well...

kellybyrd, 2017-02-03:
I can have a look at this.

kellybyrd, 2017-02-03:
I can't reproduce this with the Horus simulator, I'll try on the actual model.

kellybyrd, 2017-02-04:
I cannot reproduce this with an actual Horus radio w/ Companion 2.2 RC10 on OSX. Here is what I did:
- Powered up Hours,  I have ten models. Model4 is currently selected.
- Opened up Companion 2.2, read models from Hours. Model4 is bolded in Companion
- Changed two changes in Companion: Set Model2 as the default (bold) and changed the name of "Model7" to "_Model7". I did this as a way to confirm my changes were were written to Horus.
- From Companion, wrote models to Horus.
- Ejected mounted Hours volume from my computer
- Disconnected USB cable.

As soon as I disconnected the USB cable, the Horus switched to Model2, which was my new default. I also verified the name change to "_Model7" happened.

When I power off the Horus, and power it back on, it powers on whichever model I had loaded when I powered it off. That is the expected behavior. The "set as default" in Companion is really just "set as currently selected model" in the radio. When you select a new model in the radio before powering off, that becomes the "currently selected model" and is loaded on the next power on.

@tubeme : If I have misunderstood something please let me know. Otherwise, I think this can be closed as not a bug.

lichtl, 2017-02-05:
I am able to reproduce it on Taranis+ uploading a short video atm

tubeme, 2017-02-06:
It is exactly what you did with one little difference. I did not edit anything!.

Just read the presets from the Horus. Then write it back.

Like making a backup copy of your presets and then overwrite back the existing setup in the card.

tubeme, 2017-02-06:
Also try to choose not the first or the last model as default.

I had 5 models and the default was the third one. 

The procedure is. 

1. Choose the default model in Horus.
2. Read from TX - you should see the default model selected in Companion.
3. Write to TX.

tubeme, 2017-02-06:
In my case on Horus Power up, it chose the last model as my active profile. So theoreticaly if you choose the last profile as a default one, you will not have the jump.

projectkk2glider, 2017-02-19:
Reproduced, I will have a look.

projectkk2glider, 2017-02-19:
Perhaps I was confused, but after installing some traces I can't reproduce anymore. So we need an original Horus setup that produces this. @tubeme follow instructions here https://github.com/opentx/opentx/issues/4407#issuecomment-280909274

3djc, 2017-03-05:
Closing this as it seems to be difficult to reproduce, and a lot of work in that area as been done anyway. Please reopen if you can reproduce on RC11

closed 2017-03-05 by 3djc