opentx/opentx #4366: Q X7 LUA Rotary Left and Right events not working

gsimpson2g, 2017-02-01:
Playing around with the LUA scripting, I can't seem to get the EVT_ROT_LEFT or EVT_ROT_RIGHT events to work properly on the Q X7. All other events work, including the EVT_ENTER_BREAK when pressing the rotary down.

3djc, 2017-02-01:
I'll have a look tomorrow :)

3djc, 2017-02-01:
ok, did find some time. Those event are working perfectly both on simu and a real radio. Here is a small script you can test to demonstrate it . I'm closing this issue, please reopen if you fin contradicting evidence.
```
`x=0

local function run(event)
  lcd.clear()
	lcd.drawNumber(50,10,x)
  if event == EVT_ROT_RIGHT then
		x=x+1;
	end
	if event == EVT_ROT_LEFT then
		x=x-1;
	end
	return 0
end
return {run=run}`
```

gsimpson2g, 2017-02-01:
That's really strange. I have the latest RC10 (N362) companion software/firmware, and even the simple script didn't work in the Simu or the Radio. It remains at 0.

3djc, 2017-02-01:
I flashed again to be sure I was on N362, but yes, the script works fine here

kilrah, 2017-02-01:
What firmware options have you selected?

gsimpson2g, 2017-02-01:
I have noheli checked. Trying the script as a telemetry screen. Currently editing up a script made for the X9D+ that allows BetaFlight settings to be changed via the telemetry page with the script running on it, but can't get the rotary working.

gsimpson2g, 2017-02-01:
Ok, if I execute the script directly from the SD card, then it works 100% fine. When running the script as a Display Page, the rotary doesn't work.

raphaelcoeffic, 2017-02-01:
That's a pity, as it rules it out as a telemetry script.

kilrah, 2017-02-01:
Simply a bug.

3djc, 2017-02-02:
i checked on my X9E, it does not seem that we have those event either in that context

closed 2017-02-02 by 3djc

raphaelcoeffic, 2017-02-02:
@3djc thx!

Datospel, 2017-02-02:
"Ok, if I execute the script directly from the SD card, then it works 100% fine. When running the script as a Display Page, the rotary doesn't work."

Same for me on Q X7

3djc, 2017-02-02:
It has been fixed and will be available in next RC

janekx, 2017-06-01:
Looks 2.2 final again not work !

commit e7d7f9df (2017-02-02): Fix some events not sent to LUA telemetry scripts scripts.

Thix fixes #4366