opentx/opentx #4372: opentx 2.2 RC10 Read-Write models and settings

benoit35690, 2017-02-02:
Companion 2.2 RC10
Taranis X9D
Models using multiprotocol

1) Read Model and settings
-> in Companion : the multiprotocol detail is not copied
Reproduced at least with KN, Syma and Bayang
OK with Hisky protocol

2) Write Model and settings
-> in Taranis : the multiprotol detail is not copied
-> in the selection list of protocol, junk characters are displayed. It looks like a buffer overflow.

schwabe, 2017-02-02:
I am not sure what Multiprotocol detail should be, do you mean the option setting?

For 2. Can you give a screenshot?

benoit35690, 2017-02-02:
OK I will provide photo

benoit35690, 2017-02-02:
here are some photo

[after_read_models.jpg.zip](https://github.com/opentx/opentx/files/747724/after_read_models.jpg.zip)

[after_write_models.jpg.zip](https://github.com/opentx/opentx/files/747739/after_write_models.jpg.zip)

[junk.zip](https://github.com/opentx/opentx/files/747741/junk.zip)

commit 32f7e322 (2017-02-05): [Companion] Add missing mask when reading multi protocol (fixes #4372)
closed 2017-02-05 by schwabe